openaps/openaps #3: openaps device -h:  ImportError: No module named records

scottleibrand, 2015-05-02:
pi@raspberrypi3 ~/openaps $ openaps init myopenaps
Initialized empty Git repository in /home/pi/openaps/myopenaps/.git/
[master (root-commit) 89ef0a2] initializing openaps 0.0.0
 0 files changed
 create mode 100644 openaps.ini
pi@raspberrypi3 ~/openaps $ cd myopenaps/
pi@raspberrypi3 ~/openaps/myopenaps $ ls
openaps.ini
pi@raspberrypi3 ~/openaps/myopenaps $ openaps device -h
Traceback (most recent call last):
  File "/usr/local/bin/openaps-device", line 8, in <module>
    execfile(__file__)
  File "/home/pi/openaps/bin/openaps-device", line 9, in <module>
    from openaps import devices
  File "/home/pi/openaps/openaps/devices/__init__.py", line 3, in <module>
    import add, remove, show
  File "/home/pi/openaps/openaps/devices/add.py", line 5, in <module>
    from openaps import vendors
  File "/home/pi/openaps/openaps/vendors/__init__.py", line 2, in <module>
    import dexcom, medtronic
  File "/home/pi/openaps/openaps/vendors/medtronic.py", line 8, in <module>
    from decocare import stick, session, link, commands, history
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.8-py2.7.egg/decocare/history.py", line 11, in <module>
    from records import *
ImportError: No module named records

bewest, 2015-05-02:
Progress!
Just checking: `/home/pi/openaps` is empty directory then? (Hopefully not this repo?)

This looks like a similar issue as before, should be fixed now (https://github.com/bewest/decoding-carelink/commit/76014ccbfd8c6929be39fa79dc03eb3069cb0b26).

bewest, 2015-05-02:
Try getting latest decocare sources.

bewest, 2015-05-02:
dexcom_reader was affected by similar issue.

scottleibrand, 2015-05-02:
Yep, pulling latest sources and re-running setup.py fixed it, thanks.

closed 2015-05-02 by scottleibrand