openaps/openaps #7: openaps use cgm glucose works, iter_glucose does not work

esteward, 2015-05-05:
pi@raspberrypi ~/elliottaps $ openaps use cgm iter_glucose 5
Traceback (most recent call last):
  File "/usr/local/bin/openaps-use", line 8, in <module>
    execfile(__file__)
  File "/home/pi/openaps/bin/openaps-use", line 62, in <module>
    app( )
  File "/home/pi/openaps/openaps/cli/__init__.py", line 41, in __call__
    self.run(self.args)
  File "/home/pi/openaps/bin/openaps-use", line 56, in run
    output = app(args, self)
  File "/home/pi/openaps/openaps/uses/__init__.py", line 70, in __call__
    return self.method.selected(args)(args, app)
  File "/home/pi/openaps/openaps/uses/__init__.py", line 31, in __call__
    return self.method(args, app)
  File "/home/pi/openaps/openaps/uses/use.py", line 23, in __call__
    output = self.main(args, app)
  File "/home/pi/openaps/openaps/vendors/dexcom.py", line 86, in main
    for item in self.dexcom.iter_records('EGV_DATA'):
AttributeError: 'Dexcom' object has no attribute 'iter_records'

bewest, 2015-05-05:
Most likely cause is using an older dexcom_reader.  Try using https://github.com/bewest/dexcom_reader?

bewest, 2015-05-05:
Also, I get this if the dexcom is not plugged into the PC.
Also, you might get this if not running as root, you can install a udev file I made available in dexcom_reader, which gives permission to local users of the pc to use the dexcom device without root permission.

bewest, 2015-06-16:
Any luck?

bewest, 2015-06-16:
There's been an update, the easiest way to update is: `sudo easy_install -Z openaps`.  Please let me know if it helps or not.  Thanks!

ehwest, 2016-12-21:
File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 517, in write
    raise portNotOpenError
serial.serialutil.SerialException: Attempting to use a port that is not open