openaps/openaps #8: openaps use pump iter_pump, error

esteward, 2015-05-05:
pi@raspberrypi ~/elliottaps $ openaps use pump iter_pump
No handlers could be found for logger "decocare.stick"
Traceback (most recent call last):
  File "/usr/local/bin/openaps-use", line 8, in <module>
    execfile(__file__)
  File "/home/pi/openaps/bin/openaps-use", line 62, in <module>
    app( )
  File "/home/pi/openaps/openaps/cli/__init__.py", line 41, in __call__
    self.run(self.args)
  File "/home/pi/openaps/bin/openaps-use", line 56, in run
    output = app(args, self)
  File "/home/pi/openaps/openaps/uses/__init__.py", line 70, in __call__
    return self.method.selected(args)(args, app)
  File "/home/pi/openaps/openaps/uses/__init__.py", line 31, in __call__
    return self.method(args, app)
  File "/home/pi/openaps/openaps/uses/use.py", line 23, in __call__
    output = self.main(args, app)
  File "/home/pi/openaps/openaps/vendors/medtronic.py", line 253, in main
    for rec in self.pump.model.iter_history_pages( ):
  File "/home/pi/decoding-carelink/decocare/models/__init__.py", line 63, in __call__
    for record in page:
  File "/home/pi/decoding-carelink/decocare/models/__init__.py", line 42, in download_page
    for record in self.find_records(page):
  File "/home/pi/decoding-carelink/decocare/models/__init__.py", line 111, in find_records
    records = decoder.decode( )
  File "/home/pi/decoding-carelink/decocare/history.py", line 658, in decode
    record = parse_record(self.stream, B, larger=larger, model=self.model)
  File "/home/pi/decoding-carelink/decocare/history.py", line 593, in parse_record
    record.parse( head + date + body )
  File "/home/pi/decoding-carelink/decocare/records/base.py", line 66, in parse
    return self.decode( )
  File "/home/pi/decoding-carelink/decocare/history.py", line 280, in decode
    return dict(stale=decode_wizard_settings(stale)
  File "/home/pi/decoding-carelink/decocare/history.py", line 308, in decode_wizard_settings
    , bg_targets=decode_bg_targets(bg_targets, bg_units)
  File "/home/pi/decoding-carelink/decocare/history.py", line 344, in decode_bg_targets
    (offset, low, high) = data[start:end]
ValueError: need more than 2 values to unpack

scottleibrand, 2015-05-05:
I was able to reproduce this running iter_pump against Elliott's pump from my Pi, so it looks like this is something specific to the records on his newly set up and not-yet-in-use pump.

esteward, 2015-05-05:
i was able to work around this error by entering many temp basal events, likely to get me to 100 records

bewest, 2015-05-05:
It'd be great to use `mm-send-comm.py tweak ReadHistoryData --save --page 0`, for bunch of pages, and make sure I get the `.data` files.  What model pump?

bewest, 2015-05-05:
From the traceback, looks like there's an issue decoding the bolus wizard settings changed target.

bewest, 2015-06-16:
Closing for now, but re-open on decoding-carelink if you run into this again!  There have been a few improvements to decocare which may (or may not) have helped with this.

closed 2015-06-16 by bewest