openaps/openaps #10: Openaps use pump read_clock errors out intertmittently

amazaheri, 2015-06-13:
Openaps use pump read_clock errors out intermittently.

amazaheri, 2015-06-16:
Still seeing this with 722 1.2A

bewest, 2015-06-18:
Possibly related to https://github.com/openaps/openaps/issues/17?

FWIW, I just ran `openaps use pump read_clock` 500 times in a row without any problems:
```bash
bewest@hither:~/Documents/foo$ for x in $(seq 1 500) ; do echo $x; openaps use pump read_clock ; sleep 1;  done ; date

```

There's some logic in openaps that keeps track of how long since the last `PowerControl` dance was done, so every once in awhile, the task will take ~20 seconds longer than usual as openaps tries to negotiate a new RF session.  Here's an example, we can see it only takes around 3 seconds, but there's one in there with a 20 second offset.  This is because openaps is negotiating a new RF session behind the scenes.:

```
"2015-06-18T16:25:32"434
"2015-06-18T16:25:36"435
"2015-06-18T16:25:57"436  <-- new RF session
"2015-06-18T16:26:01"437
"2015-06-18T16:26:04"438
"2015-06-18T16:26:08"439
"2015-06-18T16:26:11"440

```

```
bewest@hither:~/Documents/foo$ python -m decocare.stick $(python -m decocare.scan) | tail -n 20
0028   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0030   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0038   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
INFO:__main__:quit send_force_read, found len: 64 expected 64 after 0 attempts
INFO:__main__:finished processing RadioStats:0x05 0x00, {'errors.timeouts': 3, 'packets.transmit': 1594L, 'errors.naks': 0, 'errors.sequence': 0, 'packets.received': 1591L, 'errors.crc': 0}
INFO:__main__:PROCESS:END:1059.113:RadioStats:0x05 0x00
INFO:__main__:INTERFACE STATS:
{'radio': {'errors.crc': 0,
           'errors.naks': 0,
           'errors.sequence': 0,
           'errors.timeouts': 3,
           'packets.received': 1591L,
           'packets.transmit': 1594L},
 'usb': {'errors.crc': 0,
         'errors.naks': 0,
         'errors.sequence': 0,
         'errors.timeouts': 0,
         'packets.received': 7482L,
         'packets.transmit': 7482L}}
INFO:__main__:howdy! all done looking at the stick
bewest@hither:~/Documents/foo$

```

Checking stick again, just to see if the counters are sane.
```
bewest@hither:~/Documents/foo$ python -m decocare.stick $(python -m decocare.scan) | tail -n 16
INFO:__main__:finished processing RadioStats:0x05 0x00, {'errors.timeouts': 3, 'packets.transmit': 1594L, 'errors.naks': 0, 'errors.sequence': 0, 'packets.received': 1591L, 'errors.crc': 0}
INFO:__main__:PROCESS:END:1058.548:RadioStats:0x05 0x00
INFO:__main__:INTERFACE STATS:
{'radio': {'errors.crc': 0,
           'errors.naks': 0,
           'errors.sequence': 0,
           'errors.timeouts': 3,
           'packets.received': 1591L,
           'packets.transmit': 1594L},
 'usb': {'errors.crc': 0,
         'errors.naks': 0,
         'errors.sequence': 0,
         'errors.timeouts': 0,
         'packets.received': 7505L,
         'packets.transmit': 7505L}}
INFO:__main__:howdy! all done looking at the stick
bewest@hither:~/Documents/foo$ 

```

HTH.

amazaheri, 2015-06-18:
Getting close, so I ran the same and executed for 285 times I got the following error every once in a while and eventually back to back:

"2015-06-18T16:09:17"182
Traceback (most recent call last):
  File "/usr/local/bin/openaps-use", line 5, in <module>
    pkg_resources.run_script('openaps==0.0.2', 'openaps-use')
  File "build/bdist.linux-armv7l/egg/pkg_resources.py", line 492, in run_script
    """Find a distribution matching requirement `req`
  File "build/bdist.linux-armv7l/egg/pkg_resources.py", line 1357, in run_script
    if sys.version_info <= (3,):
  File "build/bdist.linux-armv7l/egg/pkg_resources.py", line 47, in exec_
    except NameError:
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.2-py2.7.egg/EGG-INFO/scripts/o                                                                    penaps-use", line 63, in <module>

  File "build/bdist.linux-armv7l/egg/openaps/cli/__init__.py", line 44, in __call__
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.2-py2.7.egg/EGG-INFO/scripts/o                                                                    penaps-use", line 57, in run

  File "build/bdist.linux-armv7l/egg/openaps/uses/__init__.py", line 101, in __call__
  File "build/bdist.linux-armv7l/egg/openaps/uses/__init__.py", line 31, in __call__
  File "build/bdist.linux-armv7l/egg/openaps/uses/use.py", line 26, in __call__
  File "build/bdist.linux-armv7l/egg/openaps/vendors/medtronic.py", line 188, in main
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.15-py2.7.egg/decocare/models/                                                                    __init__.py", line 23, in __call__
    return types.MethodType(self.func, inst)(self.response)
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.15-py2.7.egg/decocare/models/                                                                    __init__.py", line 159, in read_clock
    clock = lib.parse.date(response.getData( ))
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.15-py2.7.egg/decocare/lib.py"                                                                    , line 102, in date
    return dateutil.parser.parse( data )
  File "/usr/local/lib/python2.7/dist-packages/python_dateutil-2.4.2-py2.7.egg/dateutil/p                                                                    arser.py", line 1008, in parse
    return DEFAULTPARSER.parse(timestr, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/python_dateutil-2.4.2-py2.7.egg/dateutil/p                                                                    arser.py", line 404, in parse
    ret = default.replace(**repl)
ValueError: year is out of range
183
Traceback (most recent call last):
  File "/usr/local/bin/openaps-use", line 5, in <module>
    pkg_resources.run_script('openaps==0.0.2', 'openaps-use')
  File "build/bdist.linux-armv7l/egg/pkg_resources.py", line 492, in run_script
    """Find a distribution matching requirement `req`
  File "build/bdist.linux-armv7l/egg/pkg_resources.py", line 1357, in run_script
    if sys.version_info <= (3,):
  File "build/bdist.linux-armv7l/egg/pkg_resources.py", line 47, in exec_
    except NameError:
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.2-py2.7.egg/EGG-INFO/scripts/o                                                                    penaps-use", line 63, in <module>

  File "build/bdist.linux-armv7l/egg/openaps/cli/__init__.py", line 44, in __call__
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.2-py2.7.egg/EGG-INFO/scripts/o                                                                    penaps-use", line 57, in run

  File "build/bdist.linux-armv7l/egg/openaps/uses/__init__.py", line 101, in __call__
  File "build/bdist.linux-armv7l/egg/openaps/uses/__init__.py", line 31, in __call__
  File "build/bdist.linux-armv7l/egg/openaps/uses/use.py", line 26, in __call__
  File "build/bdist.linux-armv7l/egg/openaps/vendors/medtronic.py", line 188, in main
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.15-py2.7.egg/decocare/models/                                                                    __init__.py", line 23, in __call__
    return types.MethodType(self.func, inst)(self.response)
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.15-py2.7.egg/decocare/models/                                                                    __init__.py", line 159, in read_clock
    clock = lib.parse.date(response.getData( ))
  File "/usr/local/lib/python2.7/dist-packages/decocare-0.0.15-py2.7.egg/decocare/lib.py"                                                                    , line 102, in date
    return dateutil.parser.parse( data )
  File "/usr/local/lib/python2.7/dist-packages/python_dateutil-2.4.2-py2.7.egg/dateutil/p                                                                    arser.py", line 1008, in parse
    return DEFAULTPARSER.parse(timestr, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/python_dateutil-2.4.2-py2.7.egg/dateutil/p                                                                    arser.py", line 404, in parse
    ret = default.replace(**repl)
ValueError: year is out of range
184
Traceback (most recent call last):
  File "/usr/local/bin/openaps-use", line 5, in <module>
    pkg_resources.run_script('openaps==0.0.2', 'openaps-use')
  File "build/bdist.linux-armv7l/egg/pkg_resources.py", line 492, in run_script
[... cut ...]
  File "/usr/local/lib/python2.7/dist-packages/python_dateutil-2.4.2-py2.7.egg/dateutil/p                                                                    arser.py", line 1008, in parse
    return DEFAULTPARSER.parse(timestr, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/python_dateutil-2.4.2-py2.7.egg/dateutil/p                                                                    arser.py", line 404, in parse
    ret = default.replace(**repl)
ValueError: year is out of range
187
"2015-06-18T16:10:17"188
"2015-06-18T16:10:22"189

amazaheri, 2015-06-26:
Not an issue on Rasbian, closing this.

closed 2015-06-26 by amazaheri