openaps/openaps #11: openaps use pump read_selected_basal_profile / read_basal_profile_std/ read_basal_profile_A/ read_basal_profile_B

amazaheri, 2015-06-13:
ubuntu@ubuntu:~/openaps.alimaz$ openaps use pump read_selected_basal_profile
Traceback (most recent call last):
  File "/usr/local/bin/openaps-use", line 10, in <module>
    execfile(__file__)
  File "/home/ubuntu/openaps/bin/openaps-use", line 63, in <module>
    app( )
  File "/home/ubuntu/openaps/openaps/cli/__init__.py", line 44, in __call__
    self.run(self.args)
  File "/home/ubuntu/openaps/bin/openaps-use", line 57, in run
    output = app(args, self)
  File "/home/ubuntu/openaps/openaps/uses/__init__.py", line 101, in __call__
    return self.method.selected(args)(args, app)
  File "/home/ubuntu/openaps/openaps/uses/__init__.py", line 31, in __call__
    return self.method(args, app)
  File "/home/ubuntu/openaps/openaps/uses/use.py", line 26, in __call__
    output = self.main(args, app)
  File "/home/ubuntu/openaps/openaps/vendors/medtronic.py", line 193, in main
    return getattr(self.pump.model, name)(**self.get_params(args))
AttributeError: 'PumpModel' object has no attribute 'read_selected_basal_profile'

loudnate, 2015-06-13:
What model number are you using?

amazaheri, 2015-06-13:
722

amazaheri, 2015-06-13:
v 1.2

amazaheri, 2015-06-14:
For now I can go with hard coded values but when its resolved I should be able to remove the hard coded values and update the profile dynamically @scottleibrand or alternatively get it from Mongo collection (not ideal)

amazaheri, 2015-06-14:
regardless its pure awesomeness, thank you guys!

amazaheri, 2015-06-14:
looks like original image was referencing the dev branch I reconfigured the deco using the master branch. but got the same error.

amazaheri, 2015-06-14:
I think we found a 523 model and I may switch to that instead of 722 :)

scottleibrand, 2015-06-16:
Did aliasing model numbers fix this for you?

Once we have it working, I assume you're planning to populate the profile information directly from what's on the pump, right?

bewest, 2015-06-16:
We ended up listing full classes, it's in `bewest/dev` branch.  I'd like to get carb ratios sorted out, and issue a release.

bewest, 2015-06-16:
This should be fixed, I just updated 0.0.2.  To update, `sudo easy_install -Z openaps`

closed 2015-06-16 by bewest

channemann, 2015-06-16:
If I'm getting the following, does that indicate that it's not downloading the most recent (0.0.2) and is instead simply trying to reinstall 0.0.1?

```
pi@raspberrypi ~ $ sudo easy_install -Z openaps
Searching for openaps
Best match: openaps 0.0.1
Processing openaps-0.0.1-py2.7.egg
openaps 0.0.1 is already the active version in easy-install.pth
Installing openaps-install-udev-rules script to /usr/local/bin
Installing openaps-report script to /usr/local/bin
Installing openaps script to /usr/local/bin
Installing openaps-get script to /usr/local/bin
Installing openaps-vendor script to /usr/local/bin
Installing openaps-suggest script to /usr/local/bin
Installing openaps-use script to /usr/local/bin
Installing openaps-enact script to /usr/local/bin
Installing openaps-device script to /usr/local/bin
Installing git-openaps-init script to /usr/local/bin
Using /usr/local/lib/python2.7/dist-packages/openaps-0.0.1-py2.7.egg
Processing dependencies for openaps
Finished processing dependencies for openaps
```

bewest, 2015-06-16:
Ah: `sudo easy_install -ZU openaps`.  The `-U` means "upgrade".

channemann, 2015-06-16:
I think the issue may be on my end, actually; it looks like I'm not able to access the download. Most of what I've read on that error indicates an issue with a proxy server, but I don't believe I'm behind one. I'll do a little poking around to see what I can find.
```
pi@raspberrypi ~ $ sudo easy_install -ZU openaps
Searching for openaps
Reading https://pypi.python.org/simple/openaps/
Download error on https://pypi.python.org/simple/openaps/: [Errno -2] Name or service not known -- Some packages may not be found!
Couldn't retrieve index page for 'openaps'
Scanning index of all packages (this may take a while)
Reading https://pypi.python.org/simple/
Download error on https://pypi.python.org/simple/: [Errno -2] Name or service not known -- Some packages may not be found!
Best match: openaps 0.0.1
Processing openaps-0.0.1-py2.7.egg
openaps 0.0.1 is already the active version in easy-install.pth
Installing openaps-install-udev-rules script to /usr/local/bin
Installing openaps-report script to /usr/local/bin
Installing openaps script to /usr/local/bin
Installing openaps-get script to /usr/local/bin
Installing openaps-vendor script to /usr/local/bin
Installing openaps-suggest script to /usr/local/bin
Installing openaps-use script to /usr/local/bin
Installing openaps-enact script to /usr/local/bin
Installing openaps-device script to /usr/local/bin
Installing git-openaps-init script to /usr/local/bin

Using /usr/local/lib/python2.7/dist-packages/openaps-0.0.1-py2.7.egg
Processing dependencies for openaps
Finished processing dependencies for openaps
```

amazaheri, 2015-06-16:
@scottleibrand that is correct, I am changing the logic to get as much information from Pump as possible

amazaheri, 2015-06-16:
Things on my list: selected basal profile, so we can pick the right rate based on time of the day. BG targets, Max Basal, ISF good stuff!