openaps/openaps #13: openaps use pump read_carb_ratios output some hex (expected?)

amazaheri, 2015-06-13:
ubuntu@ubuntu:~/openaps.alimaz$ openaps use pump read_carb_ratios
0000   0x00 0x14 0x00 0x00 0x00 0x00 0x00 0x00    ........
0008   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0000   0x00 0x14 0x00 0x00 0x00 0x00 0x00 0x00    ........
0008   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0000   0x00 0x14 0x00 0x00 0x00 0x00 0x00 0x00    ........
0008   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0000   0x00 0x14 0x00 0x00 0x00 0x00 0x00 0x00    ........
0008   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
{
  "units": "grams",
  "first": 1,
  "schedule": [
    {
      "i": 20,
      "x": 0,
      "r": 0,
      "ratio": 0,
      "offset": 600
    },
    {
      "i": 0,
      "x": 1,
      "r": 0,
      "ratio": 0,
      "offset": 0
    },
    {
      "i": 0,
      "x": 2,
      "r": 0,
      "ratio": 0,
      "offset": 0
    },
    {
      "i": 0,
      "x": 3,
      "r": 0,
      "ratio": 0,
      "offset": 0
    },
    {
      "i": 0,
      "x": 4,
      "r": 0,
      "ratio": 0,
      "offset": 0
    },
    {
      "i": 0,
      "x": 5,
      "r": 0,
      "ratio": 0,
      "offset": 0
    },
    {
      "i": 0,
      "x": 6,
      "r": 0,
      "ratio": 0,
      "offset": 0
    }
  ]

bewest, 2015-06-13:
Next version of decocare will fix this, there's a PR from @loudnate to fix. :-)

amazaheri, 2015-06-13:
BIG LIKE!

bewest, 2015-06-16:
The latest version has shipped, `sudo easy_install -Z openaps` should bring everything up to date.

bewest, 2015-06-16:
Thanks for catching this and other issues!

closed 2015-06-16 by bewest

amazaheri, 2015-06-16:
Thank you guys for fixing stuff :+1: