openaps/openaps #16: Results for Carb Ratios, Sensitivities, BG Targets

amazaheri, 2015-06-18:
Can we only get the items with values like the temp basal profile? right now for Carb Ratio, BG Targets and ISF we get an array with empty items:

{
  "units": "grams", 
  "first": 1, 
  "schedule": [
    {
      "i": 0, 
      "x": 0, 
      "r": 20, 
      "ratio": 20, 
      "offset": 0
    }, 
    {
      "i": 12, 
      "x": 1, 
      "r": 22, 
      "ratio": 22, 
      "offset": 360
    }, 
    {
      "i": 18, 
      "x": 2, 
      "r": 20, 
      "ratio": 20, 
      "offset": 540
    }, 
    {
      "i": 22, 
      "x": 3, 
      "r": 22, 
      "ratio": 22, 
      "offset": 660
    }, 
    {
      "i": 40, 
      "x": 4, 
      "r": 20, 
      "ratio": 20, 
      "offset": 1200
    }, 
    {
      "i": 0, 
      "x": 5, 
      "r": 0, 
      "ratio": 0, 
      "offset": 0
    }, 
    {
      "i": 0, 
      "x": 6, 
      "r": 0, 
      "ratio": 0, 
      "offset": 0
    }, 
    {
      "i": 0, 
      "x": 7, 
      "r": 0, 
      "ratio": 0, 
      "offset": 0
    }
  ]
}

bewest, 2015-06-18:
On the `dev` of decocare it now looks like this (I think this is what you meant, but not sure):

```bash
bewest@hither:~/Documents/foo$ openaps use  pump read_carb_ratios 
{
  "units": "grams", 
  "first": 1, 
  "schedule": [
    {
      "start": "00:00:00", 
      "r": 13, 
      "ratio": 13, 
      "offset": 0, 
      "i": 0, 
      "x": 0
    }, 
    {
      "start": "06:30:00", 
      "r": 15, 
      "ratio": 15, 
      "offset": 390, 
      "i": 13, 
      "x": 1
    }
  ]
}
```

```bash
bewest@hither:~/Documents/foo$ openaps use  pump read_bg_targets 
{
  "units": "mg/dL", 
  "targets": [
    {
      "high": 125, 
      "start": "00:00:00", 
      "low": 106, 
      "offset": 0, 
      "i": 0, 
      "x": 0
    }
  ], 
  "first": 1
}
```

```bash
bewest@hither:~/Documents/foo$ openaps use  pump read_insulin_sensitivies 
{
  "sensitivities": [
    {
      "i": 0, 
      "start": "00:00:00", 
      "sensitivity": 45, 
      "offset": 0, 
      "x": 0
    }
  ], 
  "action_type": "Fast", 
  "first": 1
}
```

* [ ] includes time of day spec
* [ ] don't include null/empty values

amazaheri, 2015-06-26:
Resolved in latest build. Thank you Ben!

closed 2015-06-26 by amazaheri