openaps/openaps #17: CareLink goes wild

amazaheri, 2015-06-18:
I just noticed when you hit CareLink continuously for extended time (every 5 min) it stops responding and you need to re insert it back to have things working back.

Is this common? Ben, any log I can provide to further troubleshoot? I created a brand new repo using init but still see the same thing. Ubuntu? image issue?

Same challenge with read_clock also still there.

bewest, 2015-06-18:
Not sure I understand, I can use it for hours/days without issues.

I do use these two commands to check on status of things:

* `python -m decocare.scan` - this should print the name of the serial device, should probably print something like: `/dev/serial/by-id/usb-0a21_8001-if00-port0`
* `python -m decocare.stick $(python -m decocare.scan)` - this debugs the stick status/diagnostics, it should print status of packet counts at beginning and end.  As long as the packet counts are incrementing in predictable ways, everything should be ok.

If the two previous commands work ok, I'd expect openaps to work as expected.  If the counters aren't behaving properly, or if it decocare complains about the stick not working, generally, removing the stick, and plugging in again resets it.  However, I've worked pretty hard to try and avoid putting the stick into any error state. :frowning: 
When things stop responding, it might be interesting to look at the bottom of the `/var/log/syslog` file to see if there are any errors listed.  Sorry for the hassle, hope we can get it sorted out.

bewest, 2015-06-18:
By chance, are you running more than one process trying to use the stick at the same time?  That usually messes things up in kind of the way you described.

amazaheri, 2015-06-18:
so it failed at 3:25:04

Jun 18 15:25:04 ubuntu Link opened serial port: Serial<id=0x7664b250, open=True>(port='/dev/serial/by-id/usb-0a21_8001-if00-port0', baudrate=9600, bytesize=8, parity='N', stopbits=1, timeout=0.5, xonxoff=False, rtscts=False, dsrdtr=False)
Jun 18 15:25:04 ubuntu PROCESS:OPEN:0.087 
Jun 18 15:25:04 ubuntu PROCESS:START:0.695:ProductInfo:0x04 
Jun 18 15:25:04 ubuntu link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[<ProductInfo:0x04:size(64)>] processing ProductInfo:0x04)
Jun 18 15:25:04 ubuntu send_force_read: attempt 0/2 send command, read until we get something within some timeout
Jun 18 15:25:04 ubuntu link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[<ProductInfo:0x04:size(64)>] sending ProductInfo:0x04)
Jun 18 15:25:05 ubuntu zero length READ, try once more sleep .250
Jun 18 15:25:05 ubuntu quit send_force_read, found len: 64 expected 64 after 0 attempts
Jun 18 15:25:05 ubuntu finished processing ProductInfo:0x04, {'description': 'ComLink II', 'software.version': '1.16', 'interfaces': [(0, 'Paradigm RF'), (1, 'USB')], 'product.version': '0.0', 'rf.freq': '916.5Mhz', 'serial': '0a1400'}
Jun 18 15:25:05 ubuntu PROCESS:END:1010.349:ProductInfo:0x04 
Jun 18 15:25:05 ubuntu {'description': 'ComLink II', 'software.version': '1.16', 'interfaces': [(0, 'Paradigm RF'), (1, 'USB')], 'product.version': '0.0', 'rf.freq': '916.5Mhz', 'serial': '0a1400'}
Jun 18 15:25:05 ubuntu PROCESS:START:1012.132:ProductInfo:0x04 
Jun 18 15:25:05 ubuntu link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[<ProductInfo:0x04:size(64)>] processing ProductInfo:0x04)
Jun 18 15:25:05 ubuntu send_force_read: attempt 0/2 send command, read until we get something within some timeout
Jun 18 15:25:05 ubuntu link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[<ProductInfo:0x04:size(64)>] sending ProductInfo:0x04)
Jun 18 15:25:05 ubuntu quit send_force_read, found len: 64 expected 64 after 0 attempts
Jun 18 15:25:05 ubuntu finished processing ProductInfo:0x04, {'description': 'ComLink II', 'software.version': '1.16', 'interfaces': [(0, 'Paradigm RF'), (1, 'USB')], 'product.version': '0.0', 'rf.freq': '916.5Mhz', 'serial': '0a1400'}
Jun 18 15:25:05 ubuntu PROCESS:END:1020.075:ProductInfo:0x04 
Jun 18 15:25:05 ubuntu {'description': 'ComLink II', 'software.version': '1.16', 'interfaces': [(0, 'Paradigm RF'), (1, 'USB')], 'product.version': '0.0', 'rf.freq': '916.5Mhz', 'serial': '0a1400'}
Jun 18 15:25:05 ubuntu get signal strength of Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[<ProductInfo:0x04:size(64)>]
Jun 18 15:25:05 ubuntu PROCESS:START:1022.286:SignalStrength:0x06 0x00
Jun 18 15:25:05 ubuntu link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[<SignalStrength:0x06 0x00:size(64)>] processing SignalStrength:0x06 0x00)
Jun 18 15:25:05 ubuntu send_force_read: attempt 0/2 send command, read until we get something within some timeout
Jun 18 15:25:05 ubuntu link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[<SignalStrength:0x06 0x00:size(64)>] sending SignalStrength:0x06 0x00)
Jun 18 15:25:05 ubuntu quit send_force_read, found len: 64 expected 64 after 0 attempts
Jun 18 15:25:05 ubuntu <SignalStrength:0x06 0x00:size(64)>:readSignalStrength:192
Jun 18 15:25:05 ubuntu finished processing SignalStrength:0x06 0x00, 192
Jun 18 15:25:05 ubuntu PROCESS:END:1030.544:SignalStrength:0x06 0x00
Jun 18 15:25:05 ubuntu we seem to have found a nice signal strength of: 192
Jun 18 15:25:05 ubuntu setting up to talk with 399592
Jun 18 15:25:05 ubuntu PROCESS:START:1032.549:UsbStats:0x05 0x01
Jun 18 15:25:05 ubuntu link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[<UsbStats:0x05 0x01:size(64)>] processing UsbStats:0x05 0x01)
Jun 18 15:25:05 ubuntu send_force_read: attempt 0/2 send command, read until we get something within some timeout
Jun 18 15:25:05 ubuntu link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[<UsbStats:0x05 0x01:size(64)>] sending UsbStats:0x05 0x01)
Jun 18 15:25:05 ubuntu quit send_force_read, found len: 64 expected 64 after 0 attempts
Jun 18 15:25:05 ubuntu finished processing UsbStats:0x05 0x01, {'errors.timeouts': 0, 'packets.transmit': 15449L, 'errors.naks': 0, 'errors.sequence': 0, 'packets.received': 15449L, 'errors.crc': 0}
Jun 18 15:25:05 ubuntu PROCESS:END:1040.14:UsbStats:0x05 0x01
Jun 18 15:25:05 ubuntu PROCESS:START:1040.863:RadioStats:0x05 0x00
Jun 18 15:25:05 ubuntu link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[<RadioStats:0x05 0x00:size(64)>] processing RadioStats:0x05 0x00)
Jun 18 15:25:05 ubuntu send_force_read: attempt 0/2 send command, read until we get something within some timeout
Jun 18 15:25:05 ubuntu link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[<RadioStats:0x05 0x00:size(64)>] sending RadioStats:0x05 0x00)
Jun 18 15:25:05 ubuntu quit send_force_read, found len: 64 expected 64 after 0 attempts
[... cut ...]
Jun 18 15:25:11 ubuntu send_force_read: attempt 0/2 send command, read until we get something within some timeout
Jun 18 15:25:11 ubuntu link Stick transmit[TransmitPacket:ReadPumpModel:data:unknown] reader[ReadRadio:size:0] download_i[1] status[<LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)>] poll_size[0] poll_i[4] command[<LinkStatus:0x03:status:size=64:size(64)>] sending LinkStatus:0x03:status:size=64)
Jun 18 15:25:11 ubuntu quit send_force_read, found len: 64 expected 64 after 0 attempts

amazaheri, 2015-06-18:
and ran successfully 5 min later

amazaheri, 2015-06-18:
looks like it couldn't get the pump model:

Jun 18 15:25:10 ubuntu PROCESS:START:6157.885:LinkStatus:0x03:status:size=64 
 Jun 18 15:25:10 ubuntu link Stick transmit[TransmitPacket:ReadPumpModel:data:unknown] reader[ReadRadio:size:0] download_i[1] status[LinkStatus:0x03:status:size=??LinkStatus:error:True:reason:[]:size(0)] poll_size[0] poll_i[2] command[LinkStatus:0x03:status:size=64:size(64)] processing LinkStatus:0x03:status:size=64)

bewest, 2015-06-19:
I don't think that's the problem, it looks like the problem is being able to use the stick at all.
My best theory is that cron is running a task every once in awhile.  Can you check if `crontab -l` has any output?

I believe you'll notice that `python -m decocare.stick $(python -m decocare.scan)` will also fail because it can't talk to the stick for some reason.  Most common cause of this is some other process using the stick at the same time.  cron is a program that runs things in the background on a schedule, `crontab -l` should print any such schedules.

bewest, 2015-06-19:
It's these lines that cause me to think this:

```
Jun 18 15:25:04 ubuntu PROCESS:OPEN:0.087 
Jun 18 15:25:04 ubuntu PROCESS:START:0.695:ProductInfo:0x04 
Jun 18 15:25:04 ubuntu link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[ProductInfo:0x04:size(64)] processing ProductInfo:0x04)
Jun 18 15:25:04 ubuntu send_force_read: attempt 0/2 send command, read until we get something within some timeout
Jun 18 15:25:04 ubuntu link Stick transmit[None] reader[None] download_i[False] status[None] poll_size[None] poll_i[None] command[ProductInfo:0x04:size(64)] sending ProductInfo:0x04)
Jun 18 15:25:05 ubuntu zero length READ, try once more sleep .250
Jun 18 15:25:05 ubuntu quit send_force_read, found len: 64 expected 64 after 0 attempts
Jun 18 15:25:05 ubuntu finished processing ProductInfo:0x04, {'description': 'ComLink II', 'software.version': '1.16', 'interfaces': [(0, 'Paradigm RF'), (1, 'USB')], 'product.version': '0.0', 'rf.freq': '916.5Mhz', 'serial': '0a1400'}
Jun 18 15:25:05 ubuntu PROCESS:END:1010.349:ProductInfo:0x04 
```

Usually `ProductInfo` comes back right away... in this case it took several attempts to get it, usually that only happens if some other process is also trying to use the stick.

amazaheri, 2015-06-19:
makes sense, in my case I run a loop.sh every 5 minutes, should we increase it to like 6 or 7 min?
*/5 * * * * sh  ~/openaps-test/loop.sh

bewest, 2015-06-19:
Ah, cool.  That's what causes it to fail.  If that script runs while you're trying to use openaps, you end up fighting for the stick, and potentially having to plug it in again.  With `crontab -e` you should be able to comment it out putting `#` before hand.  Commenting it out for now might be best bet while you are hacking/playing with the tools manually.

amazaheri, 2015-06-19:
correct, but the issue with read_clock happens when loop.sh is the only process running in background which I posted the log on #10

bewest, 2015-06-19:
The cron job doesn't know that you are logged in and using openaps.  If you are trying to use `read_clock` (or any other commands), and then `loop.sh` runs while you are using it, one (or both) of them will fail.

Will think about ways to let the cron job detect if the stick is already being used.  If you aren't depending on the cron job running, I'd delete it/comment it out for now.

amazaheri, 2015-06-19:
On it, story of a Windows guy doing Linux lol :)

bewest, 2015-06-19:
https://github.com/amazaheri/openaps-dev/pull/1 I think the change I made to the cron script here should fix it so the cron script should bail out if the stick is already in use.

Basically inserting:
```bash
numprocs=$(fuser -n file $(python -m decocare.scan) 2>&1 | wc -l)
if [[ $numprocs -gt 0 ]] ; then
  die "Already in use."
fi

```
Somewhere near the top should fix.

amazaheri, 2015-06-19:
awesome, testing. leaving Ras Pi alone so loop can run for a while. Thanks for all the help.

amazaheri, 2015-06-26:
No issues on Rasbian! Closing this.

closed 2015-06-26 by amazaheri