openaps/openaps #19: reservoir reads 0.0 on 723 2.4A

channemann, 2015-06-26:
openaps 0.0.3
Medtronic 723 v2.4A

```$ openaps use pump reservoir
0.0```

Reservoir actually at 203.5U

bewest, 2015-06-26:
Can you send me results of `mm-send-comm.py --serial xxyyzz tweak ReadRemainingInsulin`, the data file that results from adding `--save` might be interesting too.

channemann, 2015-07-07:
Sorry it took me awhile; here's what I get (actual reservoir is 58.0 U):

```
pi@raspberrypi ~/hannemannopenaps $ mm-send-comm.py --serial 345192 tweak ReadRemainingInsulin --save
## do stuff with an insulin pump over RF
using ` Namespace(autoinit=False, begin=None, bytesPerRecord=None, command='tweak', descr=None, dryrun=False, effectTime=None, end=None, init=False, maxRecords=None, name=None, no_postlude=False, no_prelude=False, no_rf_prelude=False, other='ReadRemainingInsulin', page=None, params=None, port='', postfix=None, prefix=None, prefix_path='', save=True, saveall=False, serial='345192', session_life=10, verbose=None) `
```
```
```
```
```
```
```
```
```javascript
{'radio': {'errors.crc': 0,
           'errors.naks': 0,
           'errors.sequence': 0,
           'errors.timeouts': 0,
           'packets.received': 7L,
           'packets.transmit': 7L},
 'usb': {'errors.crc': 0,
         'errors.naks': 0,
         'errors.sequence': 0,
         'errors.timeouts': 0,
         'packets.received': 33L,
         'packets.transmit': 33L}}
```
```
```
### PUMP MODEL: `ReadPumpModel:size[64]:data:'723'`
<class 'decocare.commands.ReadRemainingInsulin'> {}
response: ReadRemainingInsulin:size[64]:data:0.0
hexdump:
```python
0000   0x00 0x00 0x09 0x14 0x00 0x00 0x00 0x00    ........
0008   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0010   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0018   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0020   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0028   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0030   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0038   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
```
#### decoded:
```python
0.0
```
### end stats
```
```
```javascript
{'radio': {'errors.crc': 0,
           'errors.naks': 0,
           'errors.sequence': 0,
           'errors.timeouts': 0,
           'packets.received': 9L,
           'packets.transmit': 9L},
 'usb': {'errors.crc': 0,
         'errors.naks': 0,
         'errors.sequence': 0,
         'errors.timeouts': 0,
         'packets.received': 41L,
         'packets.transmit': 41L}}
```
```

Is the --save file

channemann, 2015-07-07:
After reviewing it, the data seems to be there:

`0000   0x00 0x00 0x09 0x14 0x00 0x00 0x00 0x00`

So that is (9 x 256 + 20 x 1) / 40 = 58.1. I'm assuming the 40 is from the motor step size (0.025 U minimum delivery step). Nate and I checked it out on his 723 as well and at a few different reservoir values and they are consistent.

bewest, 2015-07-13:
Nice work all!

Hmm, Might make sense to do something like this:

```python
class ReadRemainingInsulin (PumpCommand):
   ...

class ReadRemainingInsulin723 (ReadRemainingInsulin):
  basalStrokes = 40

```

Then in models.py we can have it use the right "command."  Thoughts?

channemann, 2015-07-13:
Would that take care of the byte location discrepancy as well?

channemann, 2015-07-15:
[sorry, a bit new at this]

I tried changing commands.py to this:

```
class ReadRemainingInsulin(PumpCommand):
  """
  """

  code = 115
  descr = "Read Remaining Insulin"
  params = [ ]
  retries = 2
  maxRecords = 1
  basalStrokes = 10.0
  startByte = 0
  endByte = 2
  
  def getData(self):
    data = self.data
    log.info("READ remaining insulin:\n%s" % lib.hexdump(data))
    return lib.BangInt(data[startByte:endByte])/basalStrokes

class ReadRemainingInsulin523(ReadRemainingInsulin):
  """
  """
  
  basalStrokes = 40.0
  startByte = 2
  endByte = 4
```

and models/__init__.py to this:
```
class Model523 (Model522):
  strokes_per_unit = 40
  larger = True
  read_carb_ratios = Task(commands.ReadCarbRatios)
  read_reservoir = Task(commands.ReadRemainingInsulin523)
```

Is that what you had in mind? I'm struggling to get it to work, but I'm not sure if that is because I'm not installing it properly or because the code above is simply wrong.

Also, we still need verification that the 523 and 723 act similarly, but I rolled it into the 523 for the time being (which the 723 inherits from).

bewest, 2015-07-15:
Yeah, this is basically what I had in mind.

Feel free to start a new branch: `git checkout -b wip/channeman/read_reservoir` to get a new branch.  Check in your changes, and push them to your fork: `git push origin wip/channeman/read_reservoir`.

What exactly is not working?  As you point out, the benefit of doing it this way is all the models inheriting will use the newer version you're providing.
AH:

Try this line:

```
  return lib.BangInt(data[startByte:endByte])/basalStrokes
```
to:

```
  return lib.BangInt(data[startByte:endByte])/self.basalStrokes
```

and rename `strokes_per_unit` to `basalStrokes`.  The name has to be the same.

HTH!  If you check in and push to github, I can do per-line comments to help. :-)

bewest, 2015-07-15:
Same thing for the other values... in the "class definition", `foo=bar` means a class level variable.  In the method, to access "this instances" version of the property, we need to do `self.foo`, so same thing for `self.startByte`, and `self.endByte`.

channemann, 2015-07-29:
Fixed and tested on my 723 v2.4A: wip/channeman/read_reservoir. PR submitted to dev branch of decoding-carelink (https://github.com/bewest/decoding-carelink/pull/120).

Thanks for all the assistance with basic git help, @loudnate and @bewest!

(btw, I copied your line above directly, so "channeman" is spelled incorrectly in the branch—should have four n's).

closed 2015-09-04 by channemann