openaps/openaps #21: Send_Temp_basal command

amazaheri, 2015-06-26:
Any chance if we can return a default JSON when we pass an empty JSON or {"temp":"absolute"} ? current build throws an exception.

bewest, 2016-01-01:
Yes, this would be great for first timer, or someone who wants to get into the python code.

https://github.com/openaps/openaps/blob/master/openaps/vendors/medtronic.py#L319-L320 add error checking to these lines, something like:

```
errors = [ ]
for field in [ 'rate', 'duration' ]:
  if field not in program:
    errors.append("%s is missing from input" % field)
if len(errors) > 0:
  return dict(msg="NO-OP, missing input", errors=errors, **program)
```

What should happen when required inputs are missing?

bewest, 2016-05-29:
This https://github.com/openaps/openaps/blob/master/openaps/vendors/medtronic.py#L513-L521 code always generates a json object explaining what happened.  Other proposals welcome, but avoids erroring/halting the process for now. Closing in meantime.

closed 2016-05-29 by bewest