openaps/openaps #23: Reports in stdout format throw exception

loudnate, 2015-07-03:
To reproduce (history data truncated as `[...]`)
```
$ openaps report add latest_history stdout pump read_history_data 0
added pump://stdout/read_history_data/latest_history
$ openaps report invoke latest_history
pump://stdout/read_history_data/latest_history
[...]reporting latest_history
Traceback (most recent call last):
  File "/usr/local/bin/openaps-report", line 6, in <module>
    exec(compile(open(__file__).read(), __file__, 'exec'))
  File "/root/src/openaps/bin/openaps-report", line 82, in <module>
    app( )
  File "/root/src/openaps/openaps/cli/__init__.py", line 44, in __call__
    self.run(self.args)
  File "/root/src/openaps/bin/openaps-report", line 75, in run
    output = app(args, self)
  File "/root/src/openaps/openaps/cli/subcommand.py", line 50, in __call__
    return self.method.main(args, app)
  File "/root/src/openaps/openaps/reports/invoke.py", line 40, in main
    repo.index.add([report.name])
  File "/usr/local/lib/python2.7/dist-packages/GitPython-1.0.1-py2.7.egg/git/index/base.py", line 726, in add
    entries_added.extend(self._entries_for_paths(paths, path_rewriter, fprogress, entries))
  File "/usr/local/lib/python2.7/dist-packages/GitPython-1.0.1-py2.7.egg/git/util.py", line 54, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/GitPython-1.0.1-py2.7.egg/git/index/util.py", line 84, in set_git_working_dir
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/GitPython-1.0.1-py2.7.egg/git/index/base.py", line 612, in _entries_for_paths
    entries_added.append(self._store_path(filepath, fprogress))
  File "/usr/local/lib/python2.7/dist-packages/GitPython-1.0.1-py2.7.egg/git/index/base.py", line 571, in _store_path
    st = os.lstat(filepath)     # handles non-symlinks as well
OSError: [Errno 2] No such file or directory: 'latest_history'
```

bewest, 2015-07-04:
Report commands won't do stdout the way you've been doing pipes, it's a feature intended to make development/hacking easier.  The piping with stdout might work using the `process` plugin type.  This should probably throw a different error.

The purpose of the `report` system is to kind of curry the parameters for an idempotent loop in an "auditable" way.  To do that, it forces the users to save all interstitial results as a file in the reporting system.  Hope that makes sense.  Feel free to ask more questions. :)

loudnate, 2015-07-04:
OK, I suspected that might be the case.

closed 2015-07-04 by loudnate