openaps/openaps #24: Medtronic `iter_pump` by hours instead of records

loudnate, 2015-07-04:
Hey @bewest, I'd like to add the ability to call `iter_pump` with a number of hours (e.g. 4) rather than a number of records. Where would you prefer this go?

* A new plugin
* A new Medtronic command in `openaps`
* A new, optional argument in `iter_glucose`/`iter_pump` named `--unit {count,hours,minutes,days}` to change the meaning of the `count` argument

bewest, 2015-07-04:
I like it.

bewest, 2015-07-04:
I tried creating a generic iterator such that this should be relatively easy, I'd like to put this in openaps/decocare core.

loudnate, 2015-07-04:
OK, I will add it as a separate command, then, because I think that's probably simpler to use. 

Yes, the decocare cursor will make this a 5-Line change in openaps. Thanks!

loudnate, 2015-07-21:
This issue can be closed as this went in with PR #25, but `iter_glucose_hours` won't work properly until https://github.com/bewest/decoding-carelink/pull/115 is merged.

closed 2015-10-19 by scottleibrand