openaps/openaps #29: Adding Raw data to iter_glucose

amazaheri, 2015-07-12:
Would be good to have access to raw data while in ??? or calibration mode to use to calculate required temp basal

bewest, 2015-07-13:
Good idea.

channemann, 2015-07-13:
Second that this is a good idea. Does Dexcom expose that raw data through their USB interface, or is it only picked up using an xDrip directly from the transmitter RF packets?

bewest, 2016-01-01:
It's the sensor records in dexcom_reader.

bewest, 2016-01-08:
https://github.com/openaps/openaps/pull/77

Need some patches on openxshareble to clean up logging, but it works. :-)
```
bewest@bewest-MacBookPro:~/Documents/openaps$ openaps use share iter_sen
iter_sensor_hours             iter_sensor_insertions_hours
bewest@bewest-MacBookPro:~/Documents/openaps$ openaps use share iter_sensor 2
INIT WITH SERIAL SM53306551
[ /org/bluez/hci0 ]
    org.bluez.GattManager1
    org.bluez.Media1
    org.bluez.NetworkServer1
    org.bluez.LEAdvertisingManager1
    org.bluez.Adapter1
      Name = bewest-MacBookPro
      Powered = 1
      Modalias = usb:v1D6Bp0246d0524
      DiscoverableTimeout = 180
      Alias = bewest-MacBookPro
      PairableTimeout = 0
      Discoverable = 0
      Address = AC:BC:32:CD:92:22
      Discovering = 0
      Pairable = 1
      Class = 268
      UUIDs = dbus.Array([dbus.String(u'00001801-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110e-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001200-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001800-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110c-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1)
[ /org/bluez/hci0/dev_C8_69_CD_7D_90_C3 ]
    org.bluez.Device1
      Paired = 0
      Adapter = /org/bluez/hci0
      LegacyPairing = 0
      Alias = C8-69-CD-7D-90-C3
      ManufacturerData = dbus.Dictionary({dbus.UInt16(76): dbus.Array([dbus.Byte(9), dbus.Byte(6), dbus.Byte(3), dbus.Byte(3), dbus.Byte(10), dbus.Byte(0), dbus.Byte(0), dbus.Byte(23)], signature=dbus.Signature('y'), variant_level=1)}, signature=dbus.Signature('qv'), variant_level=1)
      Connected = 0
      UUIDs = dbus.Array([], signature=dbus.Signature('s'), variant_level=1)
      Address = C8:69:CD:7D:90:C3
      Trusted = 0
      Blocked = 0
[ /org/bluez/hci0/dev_D0_03_4B_EC_92_4F ]
    org.bluez.Device1
      Paired = 0
      Adapter = /org/bluez/hci0
      LegacyPairing = 0
      Alias = D0-03-4B-EC-92-4F
      ManufacturerData = dbus.Dictionary({dbus.UInt16(76): dbus.Array([dbus.Byte(9), dbus.Byte(6), dbus.Byte(3), dbus.Byte(49), dbus.Byte(10), dbus.Byte(0), dbus.Byte(0), dbus.Byte(26)], signature=dbus.Signature('y'), variant_level=1)}, signature=dbus.Signature('qv'), variant_level=1)
      Connected = 0
      UUIDs = dbus.Array([], signature=dbus.Signature('s'), variant_level=1)
      Address = D0:03:4B:EC:92:4F
      Trusted = 0
      Blocked = 0
[ /org/bluez ]
    org.bluez.ProfileManager1
    org.bluez.AgentManager1
[ /org/bluez/hci0/dev_E8_37_9C_DB_FC_BD ]
    org.bluez.Device1
      Name = DEXCOMRX
      Paired = 0
      Adapter = /org/bluez/hci0
      LegacyPairing = 0
      Alias = DEXCOMRX
      Connected = 1
      UUIDs = dbus.Array([dbus.String(u'f0aca0b1-ebfa-f96f-28da-076c35a521db')], signature=dbus.Signature('s'), variant_level=1)
      Address = E8:37:9C:DB:FC:BD
      Trusted = 0
      Blocked = 0
[ /org/bluez/hci0/dev_10_40_F3_F0_E4_ED ]
    org.bluez.Device1
      Paired = 0
      Adapter = /org/bluez/hci0
      LegacyPairing = 0
[... 31 lines cut ...]
f0acacac-ebfa-f96f-28da-076c35a521db <Adafruit_BluefruitLE.bluez_dbus.gatt.BluezGattCharacteristic object at 0x7fa285758b90> <Interface <ProxyObject wrapping <dbus._dbus.SystemBus (system) at 0x7fa276f71530> :1.1 /org/bluez/hci0/dev_E8_37_9C_DB_FC_BD/service000c/char0016 at 0x7fa285758950> implementing 'org.bluez.GattCharacteristic1' at 0x7fa285758490>
f0acb0cc-ebfa-f96f-28da-076c35a521db <Adafruit_BluefruitLE.bluez_dbus.gatt.BluezGattCharacteristic object at 0x7fa285758b50> <Interface <ProxyObject wrapping <dbus._dbus.SystemBus (system) at 0x7fa276f71530> :1.1 /org/bluez/hci0/dev_E8_37_9C_DB_FC_BD/service000c/char0018 at 0x7fa285758350> implementing 'org.bluez.GattCharacteristic1' at 0x7fa285758f10>
f0acb0cd-ebfa-f96f-28da-076c35a521db <Adafruit_BluefruitLE.bluez_dbus.gatt.BluezGattCharacteristic object at 0x7fa285758f90> <Interface <ProxyObject wrapping <dbus._dbus.SystemBus (system) at 0x7fa276f71530> :1.1 /org/bluez/hci0/dev_E8_37_9C_DB_FC_BD/service000c/char001a at 0x7fa285758990> implementing 'org.bluez.GattCharacteristic1' at 0x7fa285758fd0>
sending auth code SM53306551
<Adafruit_BluefruitLE.bluez_dbus.gatt.BluezGattCharacteristic object at 0x7fa2857d23d0>
DEXCOM <openxshareble.ble.readdata.Device object at 0x7fa2857bafd0>
_heartbeat_tick bb
[ 
  { 
    "filtered": 291200,
    "system_time": "2016-01-08T20:12:23",
    "display_time": "2016-01-08T13:12:22",
    "unfiltered": 290816,
    "rssi": 177
  },
  { 
    "filtered": 292480,
    "system_time": "2016-01-08T20:07:23",
    "display_time": "2016-01-08T13:07:22",
    "unfiltered": 290112,
    "rssi": 181
  }
]bewest@bewest-MacBookPro:~/Documents/openaps$

```

bewest, 2016-05-29:
https://github.com/openaps/dexcom_reader shipped (with a a corresponding update to openaps) to include both sensor and calibration records.  Enjoy. :smiley_cat:

closed 2016-05-29 by bewest