openaps/openaps #30: git object corruption issue

amazaheri, 2015-07-12:
reporting clock.json
Traceback (most recent call last):
 File "/usr/local/bin/openaps-report", line 5, in <module>
 pkg_resources.run_script('openaps==0.0.4', 'openaps-report')
 File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 499, in run_scr ipt
 self.require(requires)[0].run_script(script_name, ns)
 File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 1235, in run_sc ript
 execfile(script_filename, namespace, namespace)
 File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.4-py2.7.egg/EGG-INFO/ scripts/openaps-report", line 82, in <module>
 app( )
 File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.4-py2.7.egg/openaps/c li/init.py", line 45, in call
 self.epilog( )
 File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.4-py2.7.egg/EGG-INFO/ scripts/openaps-report", line 69, in epilog
 super(ReportToolApp, self).epilog( )
 File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.4-py2.7.egg/openaps/c li/init.py", line 64, in epilog
 self.create_git_commit( )
 File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.4-py2.7.egg/openaps/c li/init.py", line 72, in create_git_commit
 if self.repo.is_dirty( ) or self.repo.index.diff(None):
 File "/usr/local/lib/python2.7/dist-packages/GitPython-1.0.1-py2.7.egg/git/rep o/base.py", line 582, in is_dirty
 len(self.git.diff('--cached', default_args)):
 File "/usr/local/lib/python2.7/dist-packages/GitPython-1.0.1-py2.7.egg/git/cmd .py", line 440, in <lambda>
 return lambda args, kwargs: self._call_process(name, *args, kwargs)
 File "/usr/local/lib/python2.7/dist-packages/GitPython-1.0.1-py2.7.egg/git/cmd .py", line 834, in _call_process
 return self.execute(make_call(), **_kwargs)
 File "/usr/local/lib/python2.7/dist-packages/GitPython-1.0.1-py2.7.egg/git/cmd .py", line 627, in execute
 raise GitCommandError(command, status, stderr_value)
git.exc.GitCommandError: 'git diff --cached --abbrev=40 --full-index --raw' retu rned with exit code 128
stderr: 'error: object file .git/objects/91/cc8d05fd89c0415d34592847f917fa51f2d9 67 is empty
fatal: loose object 91cc8d05fd89c0415d34592847f917fa51f2d967 (stored in .git/obj ects/91/cc8d05fd89c0415d34592847f917fa51f2d967) is corrupt'

bewest, 2016-01-01:
Still an issue. :-(  I've used this https://gist.github.com/bewest/a88ee1c7d3b0c7c7af1b with some success.

scottleibrand, 2016-01-01:
Would that be suitable for inclusion in oref0-reset-git as a "try first"?

-Scott

bewest, 2016-03-06:
Latest attempt: https://github.com/openaps/openaps/pull/81

bewest, 2016-03-21:
That latest attempt introduced a side-effect; by switching to using the `index`, and giving up`repo.commit('-av')`, it changed the semantics in the `dev` branch.  The change in semantics means that the `-a` option was dropped, meaning changes are only recorded explicitly when added, instead of implicitly when changed.  https://github.com/openaps/openaps/commit/cdc09cc2056fad6fe1495127396f76fc9b2f5879 emulates the `-a` option, by adding any updated path to the index.

This should catch all the `add`, `remove`, and `import` utilities.  It should also restore sane behavior when changes are made (on a tracked file) and then `openaps` runs, it will then record the change along with the latest batch.