openaps/openaps #32: read_bg_targets sometimes generates dangerously incorrect results

scottleibrand, 2015-07-19:
For some reason read_bg_targets sometimes outputs results with  "units": null, and sets the OpenAPS target BG to 50 (!).

It looks like read_carb_ratios is also doing the same.

You can see more details in my openaps-dev repo.  Command I'm running is:
export file=bg_targets.json && git log $file | grep commit | awk '{print $2}' | while read hash; do git show $hash; done | less

```
commit 11a32f59627b6fcdbecb78cdb4e2ab31e683a693
Author: Scott Leibrand <scottleibrand@gmail.com>
Date:   Sun Jul 19 08:03:26 2015 -0700

    openaps-report invoke profile.json iob.json requestedtemp.json

          TODO: better change descriptions
          /usr/local/bin/openaps-report invoke profile.json iob.json requestedtemp.json

diff --git a/bg_targets.json b/bg_targets.json
index c802e19..555c074 100644
--- a/bg_targets.json
+++ b/bg_targets.json
@@ -1,14 +1,14 @@
 {
-  "units": "mg/dL",
+  "units": null,
   "targets": [
     {
-      "high": 120,
-      "start": "00:00:00",
-      "low": 115,
-      "offset": 0,
-      "i": 0,
+      "high": 50,
+      "start": "02:30:00",
+      "low": 50,
+      "offset": 1590,
+      "i": 53,
       "x": 0
     }
   ],
-  "first": 1
+  "first": 3
 }
\ No newline at end of file
```
and
```
commit c5bc3ff8d03260eb74b0228286a10650e819f595
Author: Scott Leibrand <scottleibrand@gmail.com>
Date:   Sat Jul 18 22:55:48 2015 -0700

    openaps-report invoke profile.json iob.json requestedtemp.json

          TODO: better change descriptions
          /usr/local/bin/openaps-report invoke profile.json iob.json requestedtemp.json

diff --git a/bg_targets.json b/bg_targets.json
index c802e19..f6026a3 100644
--- a/bg_targets.json
+++ b/bg_targets.json
@@ -1,14 +1,22 @@
 {
-  "units": "mg/dL",
+  "units": null,
   "targets": [
     {
-      "high": 120,
-      "start": "00:00:00",
-      "low": 115,
-      "offset": 0,
-      "i": 0,
+      "high": 7,
+      "start": "04:00:00",
+      "low": 40,
+      "offset": 1680,
+      "i": 56,
       "x": 0
+    },
+    {
+      "high": 18,
+      "start": "15:30:00",
+      "low": 7,
+      "offset": 6690,
+      "i": 223,
+      "x": 1
     }
   ],
-  "first": 1
+  "first": 21
 }
\ No newline at end of file
diff --git a/carb_ratio.json b/carb_ratio.json
index 9d5267a..e9ae799 100644
--- a/carb_ratio.json
+++ b/carb_ratio.json
@@ -1,14 +1,22 @@
 {
-  "units": "grams",
-  "first": 1,
+  "units": null,
+  "first": 3,
   "schedule": [
     {
-      "start": "00:00:00",
-      "r": 10,
-      "ratio": 10,
-      "offset": 0,
-      "i": 0,
+      "start": "02:30:00",
+      "r": 50,
+      "ratio": 50,
+      "offset": 1590,
+      "i": 53,
       "x": 0
+    },
+    {
+      "start": "01:00:00",
+      "r": 0,
+      "ratio": 0,
+      "offset": 1500,
+      "i": 50,
+      "x": 1
     }
   ]
 }
\ No newline at end of file
```

scottleibrand, 2015-07-19:
My workaround for now is https://github.com/openaps/openaps-js/commit/e9f443f3ec77f3ecf86bcfa443e6485f8c7e6138

scottleibrand, 2015-07-29:
Looks like this is a more general issue.  Just had an instance where it read all kind of random incorrect numbers for pump settings.

Would it be feasible to have openaps (or decocare) do some more stringent error-checking on things that we know should be non-zero and/or always less than a certain upper bound?

scottleibrand, 2015-07-29:
https://gist.github.com/scottleibrand/49769a94e4c6c0ce4561

loudnate, 2015-07-31:
```json
$ atomicaps use pump read_bg_targets
{
  "units": "mg/dL", 
  "raw": "0x01 0x00 0x5a 0x78 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00", 
  "targets": [
    {
      "high": 120, 
      "start": "00:00:00", 
      "low": 90, 
      "offset": 0, 
      "i": 0, 
      "x": 0
    }
  ], 
  "first": 1
}
```

scottleibrand, 2015-08-01:
What's the magic incantation again to easy_install the dev branch?

loudnate, 2015-08-01:
You'll have to check out the branch and run `python setup.py install`

bewest, 2015-08-01:
The openaps `setup.py` has the hint:

`https://github.com/bewest/decoding-carelink/tarball/dev#egg=decocare-0.0.17-dev-1`

Try: `sudo easy_install -ZU https://github.com/bewest/decoding-carelink/tarball/dev#egg=decocare-0.0.17-dev-1` to get decocare.

scottleibrand, 2015-08-02:
```
diff --git a/bg_targets.json.new b/bg_targets.json.new
index f586557..50a6efc 100644
--- a/bg_targets.json.new
+++ b/bg_targets.json.new
@@ -1,15 +1,15 @@
 {
-  "units": "mg/dL",
-  "raw": "0x01 0x00 0x6e 0x78 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x
00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00",
+  "units": null,
+  "raw": "0x34 0x00 0x00 0x28 0x00 0x05 0x22 0x00 0x09 0x2c 0x00 0x0f 0x34 0x00 0x13 0x2a 0x00 0x1a 0x24 0x00 0x1e 0x2e 0x00 0x25 0x34 0x00 0x2b 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x
00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00",
   "targets": [
     {
-      "high": 120,
+      "high": 40,
       "start": "00:00:00",
-      "low": 110,
+      "low": 0,
       "offset": 0,
       "i": 0,
       "x": 0
     }
   ],
-  "first": 1
+  "first": 52
 }
\ No newline at end of file
```

bewest, 2016-05-29:
There were other issues causing problems as well, which have been fixed.  (The bg_targets response turned out to be twice as big as originally thought.)  Believe this is fixed, but please re-open if it crops up again.

closed 2016-05-29 by bewest