openaps/openaps #35: openaps init doesn't work on CWD

scottleibrand, 2015-07-26:
```
pi@raspberrypi2:~$ cp -r openaps-dev/ openaps-test/
pi@raspberrypi2:~$ cd openaps-test/
pi@raspberrypi2:~/openaps-test$ rm -rf .git
pi@raspberrypi2:~/openaps-test$ openaps init
Initialized empty Git repository in /home/pi/openaps-test/.git/
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
pi@raspberrypi2:~/openaps-test$ cd ..
pi@raspberrypi2:~$ openaps init openaps-test/
Reinitialized existing Git repository in /home/pi/openaps-test/.git/
[master (root-commit) 55fb38f] initializing openaps 0.0.3
 1 file changed, 199 insertions(+)
 create mode 100644 openaps.ini
```

bewest, 2015-10-06:
I think you can do `openaps init .` for CWD.

scottleibrand, 2015-10-26:
It sure does.

closed 2015-10-26 by scottleibrand