openaps/openaps #36: millisecond date doesn't match timestamp in pumphistory.json

scottleibrand, 2015-07-26:
Pump time is set correctly in PDT.  timestamp is correct.  Actual current unix timestamp is:
1437948234
date record on most recent TempBasalDuration record is:
1437951103000.0
which is 1 hour ahead.

```
"2015-07-26T14:57:48"pi@rpi3 ~/openaps-test $ openaps pumpquery
pump://JSON/read_clock/clock.json
reporting clock.json
pump://JSON/read_temp_basal/currenttemp.json
reporting currenttemp.json
pump://JSON/iter_pump/pumphistory.json
reporting pumphistory.json
pi@rpi3 ~/openaps-test $ date +%s
1437948234
pi@rpi3 ~/openaps-test $ cat clock.json
"2015-07-26T15:02:36"pi@rpi3 ~/openaps-test $
pi@rpi3 ~/openaps-test $ head pumphistory.json
[
  {
    "_type": "TempBasalDuration",
    "duration (min)": 30,
    "_description": "TempBasalDuration 2015-07-26T14:51:43 head[2], body[0] op[0x16]",
    "date": 1437951103000.0,
    "timestamp": "2015-07-26T14:51:43",
    "_body": "",
    "_head": "1601",
    "_date": "6bf30e5a0f"
pi@rpi3 ~/openaps-test $
```

scottleibrand, 2015-07-26:
Changing the timezone to Alaska doesn't seem to make a difference:
```
pi@rpi3 ~/openaps-test $ sudo dpkg-reconfigure tzdata
Current default time zone: 'US/Alaska'
Local time is now:      Sun Jul 26 14:07:35 AKDT 2015.
Universal Time is now:  Sun Jul 26 22:07:35 UTC 2015.
pi@rpi3 ~/openaps-test $ openaps pumpquery
pump://JSON/read_clock/clock.json
reporting clock.json
pump://JSON/read_temp_basal/currenttemp.json
reporting currenttemp.json
pump://JSON/iter_pump/pumphistory.json
reporting pumphistory.json
pi@rpi3 ~/openaps-test $ head pumphistory.json
[
  {
    "_type": "TempBasalDuration",
    "duration (min)": 30,
    "_description": "TempBasalDuration 2015-07-26T14:51:43 head[2], body[0] op[0x16]",
    "date": 1437958303000.0,
    "timestamp": "2015-07-26T14:51:43",
    "_body": "",
    "_head": "1601",
    "_date": "6bf30e5a0f"
pi@rpi3 ~/openaps-test $ cat clock.json
"2015-07-26T15:15:48"pi@rpi3 ~/openaps-test $ date +%s
1437949042
```
But somehow CDT does?  This makes no sense...
```
Current default time zone: 'US/Central'
Local time is now:      Sun Jul 26 17:19:57 CDT 2015.
Universal Time is now:  Sun Jul 26 22:19:57 UTC 2015.

pi@rpi3 ~/openaps-test $ openaps pumpquery
pump://JSON/read_clock/clock.json
reporting clock.json
pump://JSON/read_temp_basal/currenttemp.json
reporting currenttemp.json
pump://JSON/iter_pump/pumphistory.json
reporting pumphistory.json
pi@rpi3 ~/openaps-test $ head pumphistory.json
[
  {
    "_type": "TempBasalDuration",
    "duration (min)": 30,
    "_description": "TempBasalDuration 2015-07-26T14:51:43 head[2], body[0] op[0x16]",
    "date": 1437943903000.0,
    "timestamp": "2015-07-26T14:51:43",
    "_body": "",
    "_head": "1601",
    "_date": "6bf30e5a0f"
pi@rpi3 ~/openaps-test $ date +%s
1437949306
```

loudnate, 2015-07-27:
(I'm looking into this now; this is a change in decocare – I don't have access to assign issues in this repo)