openaps/openaps #37: Prevent openaps.ini from being truncated

scottleibrand, 2015-07-27:
Occasionally my openaps.ini gets truncated to 0 bytes.  I presume that's because openaps is truncating the file and re-writing it when it updates the pump expires time. Can we change the way openaps updates it to make a copy, verify it is valid, and then do an atomic move to overwrite the new file?

loudnate, 2015-07-27:
Interesting. Have you isolated which failures trigger this case? Eg Python stacktraces?

I have a branch in my openaps repo that cuts down on the number of times the pump is queried for model number, which might help? I haven't experienced this personally.

scottleibrand, 2015-07-27:
No, I've never caught it in the act, just noticed everything is broken due
to the truncated file.  It's quite likely that it's caused by the program
getting killed (or the Pi powered off) unexpectedly, though.  What is the
mechanism by which we write to openaps.ini?  Could that be done more
atomically?

loudnate, 2015-07-27:
I'm pretty sure it's the common `with open('filename', mode='w') as fp:`
block in Python. I'm not aware of out-of-the-box atomic write flags but
your suggestion of write-to-temp-and-copy is a good one.

bewest, 2015-07-28:
`mv` is an atomic operation in linux, so it's a good idea to write elsewhere, then `mv` into place.  Good idea.

bewest, 2016-01-08:
I think this has gone away, mostly.

bewest, 2016-01-13:
Any truncation issues, I think I've seen a few of the extra inis get truncated from time to time, which is what I would expect.

scottleibrand, 2016-01-13:
I have only seen pump.ini truncation, not openaps.ini.

bewest, 2016-05-29:
This wound up shipping... should no longer see `ini` files truncated... they are never opened for writing during normal execution any more...  This wound up shipping, so going to close for now to help clear the decks, but please re-open if it appears with `openaps 0.1.1` or later.

closed 2016-05-29 by bewest