openaps/openaps #41: Add Additional Decom Functionality

channemann, 2015-09-04:
Currently, the only Dexcom vendor/device functions available are

`glucose`
`iter_glucose`
`iter_glucose_hours`

which all obtain glucose values, trend arrow direction, system times, and display times.

There are several other functions exposed by dexcom_reader, including device settings, battery status, and I believe calibration entries. It would be great to add this functionality to openaps.

bewest, 2016-01-02:
Happy new year, I think this shipped earlier today.

channemann, 2016-01-07:
Looks like the version that shipped has a reversion of some sort. `iter_calibrations` and `iter_calibrations_hours` somehow (they both return `[]`), while `calibrations` returns a truncated list (missing the most recent handful of calibrations). This is what we fixed before, but I'm not sure how it went backward in this release.

channemann, 2016-01-08:
@bewest Wondering if the issue is related to https://github.com/openaps/openaps/blob/master/setup.py#L23; should the dependency version be incremented? The issue is fixed when I switch back to https://github.com/channemann/dexcom_reader/tree/wip/fix-pages, which is apparently current with the most recent version of `openaps/dexcom_reader`, so I'm guessing that updating `openaps` and running `develop` installed an earlier version?

bewest, 2016-01-12:
How's this:

```
bewest@bewest-MacBookPro:~/Documents/openaps$ openaps use share -h
usage: openaps-use share [-h] USAGE ...

optional arguments:
  -h, --help            show this help message and exit

## Device share:
  vendor openxshareble
  
  openxshareble - pure python driver to communicate with Dexcom G4+Share
  over ble.  Allows openaps to use ble to talk to Dexcom G4+Share.
  
  Examples:
  ```
  # add vendor to openaps
  openaps vendor add openxshareble
  # create device called share to use the new vendor
  openaps device add share openxshareble
  # use it, eg:
  openaps use share iter_glucose 2
  ```
  
  
      

  USAGE                 Usage Details
    GetFirmwareHeader
    ReadBatteryLevel
    ReadBatteryState
    ReadManufacturingData
    ReadTransmitterId
    battery
    calibrations        read calibration entry records
    configure           Configure DEXCOMRX serial number.
    glucose             glucose
    insertion_time      Fetch INSERTION_TIME records from the Dexcom receiver.
    iter_calibrations   read last <count> calibration records, default 10, eg:
    iter_calibrations_hours
                        read last <hours> of calibration records, default 1,
                        eg:
    iter_glucose        read last <count> glucose records, default 100, eg:
    iter_glucose_hours  read last <hours> of glucose records, default 1, eg:
    iter_insertion_time
    iter_insertion_time_hours
    iter_meter_data
    iter_meter_data_hours
    iter_sensor
    iter_sensor_hours
    iter_sensor_insertions
                        read last <count> sensor insertion, removal, and
                        expiration records, default 10, eg:
    iter_sensor_insertions_hours
                        read last <hours> of sensor insertion, removal, and
                        expiration records, default 1, eg:
    iter_user_event_data
    iter_user_event_data_hours
    list_dexcom         Scan the environment looking for Dexcom devices.
                        Returns a list of scanned devices.
    meter_data          Fetch METER_DATA records from the Dexcom receiver.
    scan                scan for usb stick
    sensor              Fetch Sensor (raw) records from Dexcom receiver.
    sensor_insertions   read sensor insertion, removal, and expiration records
                        of sensors
    user_event_data     Fetch USER_EVENT_DATA records from the Dexcom
                        receiver.
bewest@bewest-MacBookPro:~/Documents/openaps$ 

```

channemann, 2016-01-12:
+1

Awesome stuff; would love to be able to not have my Dexcom receiver physically attached for a variety of reasons. That's all being done on your laptop at the moment, right? Have you tried it with a BT dongle in the RPi?

bewest, 2016-01-12:
It's possible to get openaps running on OSX laptop (everything except carelink, if using yardstick1, RL, or similar everything can be done from laptop).
I'm missing a BLE dongle for rpi :-(.  Believe it works out of box for edison, but not sure.

bewest, 2016-05-29:
calibrations and sensor readings just shipped, there are now a ton of uses, along with integrated G5 usb support.  Closing for now, many thanks.

closed 2016-05-29 by bewest