openaps/openaps #47: Use only one glucose measurement in json

oskarpearson, 2015-09-21:
In all the places I've found, openaps has standardised on reporting glucose levels in mg/dL

However, in https://github.com/openaps/openaps/blob/master/openaps/vendors/medtronic.py#L248-L249 it returns the data in the units specified by the user.

This means that all code that consumes the glucose targets needs to be able to handle two cases. Unfortunately, this could lead to dire consequences, where we try and get sugar levels down to 5.0 in the wrong scale. This sort of thing has happened before: http://www.wired.com/2010/11/1110mars-climate-observer-report/ 

I think we should standardise entirely on mg/dL in all presented/created/parsed json - and leave it up to tools to display the values differently on the UI if needs be. This moves it to the presentation layer, and means that problems like the above can be avoided.

https://github.com/loudnate/openaps-monitor/pull/24 is an example of this happening in the presentation layer. We can have a field that contains the user-preferred units in the settings json, for example, that would make this sort of thing easier.

If everyone is ok with this, I'd like to change the bg targets code to always return mg/dL

oskarpearson, 2015-10-01:
@bewest Not sure if yo think that this is a good idea. Let me know if I should refactor!

bewest, 2015-10-01:
I think decocare should detect the units the pump is using and report those.
Openaps can have an option to switch units if needed; the units used should be passed with the data somehow.

A use can be created to "convert $xyz data in mmol to mg/dL" to help other uses/reports if needed, the most worrying thing here is that decocare is failing to detect the units used by the pump appropriately.

closed 2015-12-15 by oskarpearson