openaps/openaps #54: Large square bolus incorrectly read

channemann, 2015-09-28:
Pump: MM 723 v2.4A
Scenario: Bolus wizard used to deliver a square bolus, 6.6 u over 30 min
Issue: `iter_pump` reads it as only 0.2 u over 30 min

Output below shows records on either side of the bolus wizard delivery. I'm guessing this is a decocare issue, possibly with an extra bit not being accounted for when larger boluses are delivered (similar to the issue with high BG values during MM calibration, perhaps).

```
pi@raspberrypi ~/hannemannopenaps $ openaps use pump iter_pump 4
[
  {
    "_type": "BasalProfileStart", 
    "_description": "BasalProfileStart 2015-09-28T12:27:37 head[2], body[3] op[0x7b]", 
    "timestamp": "2015-09-28T12:27:37", 
    "_body": "0d2400", 
    "offset": 23400000, 
    "_head": "7b02", 
    "rate": 0.9, 
    "_date": "a55b0c1c0f"
  }, 
  {
    "programmed": 0.2, 
    "_type": "Bolus", 
    "_description": "Bolus 2015-09-28T12:00:36 head[8], body[0] op[0x01]", 
    "timestamp": "2015-09-28T12:00:36", 
    "_body": "", 
    "appended": [
      {
        "_type": "UnabsorbedInsulinBolus", 
        "_description": "UnabsorbedInsulinBolus unknown head[17], body[0] op[0x5c]", 
        "data": [
          {
            "amount": 1.2, 
            "age": 152
          }, 
          {
            "amount": 0.45, 
            "age": 212
          }, 
          {
            "amount": 0.55, 
            "age": 222
          }, 
          {
            "amount": 0.55, 
            "age": 232
          }, 
          {
            "amount": 0.05, 
            "age": 242
          }
        ], 
        "_body": "", 
        "_head": "5c113098c012d4c016dec016e8c002f2c0", 
        "_date": ""
      }
    ], 
    "_head": "0101080108000801", 
    "amount": 0.2, 
    "unabsorbed": 0.2, 
    "duration": 30, 
    "type": "square", 
    "_date": "a4406c7c0f"
  }, 
  {
    "_type": "BolusWizard", 
    "bg": 0, 
    "bg_target_high": 120, 
    "_description": "BolusWizard 2015-09-28T12:00:36 head[2], body[15] op[0x5b]", 
    "timestamp": "2015-09-28T12:00:36", 
    "_body": "505000782864000108000000010878", 
    "_head": "5b00", 
    "unabsorbed_insulin_total": 0.0, 
    "correction_estimate": 0.0, 
    "_date": "a4400c7c0f", 
    "bolus_estimate": 6.6, 
    "bg_target_low": 100, 
    "carb_ratio": 8.0, 
    "food_estimate": 6.6, 
    "carb_input": 80, 
    "sensitivity": 40
  }, 
  {
    "_type": "TempBasalDuration", 
    "duration (min)": 30, 
    "_description": "TempBasalDuration 2015-09-28T11:57:37 head[2], body[0] op[0x16]", 
    "timestamp": "2015-09-28T11:57:37", 
    "_body": "", 
    "_head": "1601", 
    "_date": "a5790b5c0f"
  }, 
  {
    "_type": "TempBasal", 
    "temp": "absolute", 
    "_description": "TempBasal 2015-09-28T11:57:37 head[2], body[1] op[0x33]", 
    "timestamp": "2015-09-28T11:57:37", 
    "_body": "00", 
    "_head": "3301", 
    "rate": 0.025, 
    "_date": "a5790b5c0f"
  }
]
```

loudnate, 2015-09-28:
6.2 * 40 = 256

You know the drill! ;-)

channemann, 2015-09-29:
Close, 6.4 x 40 = 256 .... but yeah, time to go bit hunting!

channemann, 2015-09-29:
So I believe this is where the issue is located:

https://github.com/bewest/decoding-carelink/blob/master/decocare/records/bolus.py#L37-L39

I'm guessing the issue will not apply to the "smaller" pumps because of the 10 vs. 40 steps, hence never getting above 256.

Here is another square bolus that does not have the error:

```
{
    "programmed": 4.0, 
    "_type": "Bolus", 
    "_description": "Bolus 2015-09-29T13:02:02 head[8], body[0] op[0x01]", 
    "timestamp": "2015-09-29T13:02:02", 
    "_body": "", 
    "appended": [
      {
        "_type": "UnabsorbedInsulinBolus", 
        "_description": "UnabsorbedInsulinBolus unknown head[5], body[0] op[0x5c]", 
        "data": [
          {
            "amount": 5.4, 
            "age": 244
          }
        ], 
        "_body": "", 
        "_head": "5c05d8f4c0", 
        "_date": ""
      }
    ], 
    "_head": "0100a00000000001", 
    "amount": 0.0, 
    "unabsorbed": 0.0, 
    "duration": 30, 
    "type": "square", 
    "_date": "82426d7d0f"
  }, 
  {
    "_type": "BolusWizard", 
    "bg": 0, 
    "bg_target_high": 120, 
    "_description": "BolusWizard 2015-09-29T13:02:02 head[2], body[15] op[0x5b]", 
    "timestamp": "2015-09-29T13:02:02", 
    "_body": "3750007828640000b400000000b478", 
    "_head": "5b00", 
    "unabsorbed_insulin_total": 0.0, 
    "correction_estimate": 0.0, 
    "_date": "82420d7d0f", 
    "bolus_estimate": 4.5, 
    "bg_target_low": 100, 
    "carb_ratio": 8.0, 
    "food_estimate": 4.5, 
    "carb_input": 55, 
    "sensitivity": 40
  }
```

Comparing the "_head" output:

For 6.6 units: 0101080108000801 (6.6 - 6.4 = 0.2 units at 40 steps/unit = 8 = 08)
For 4.0 units: 0100a00000000001 (4.0 units at 40 steps/unit = 160 = a0)

channemann, 2015-09-29:
Should the fix just then be

`'programmed': self.head[2]/40.0 + self.head[1]*6.4,` ?

channemann, 2015-10-06:
@bewest can you get data from an x22 pump for a very large (> 6.4 u) bolus? I want to see if this is x23 models only.

channemann, 2015-10-06:
@bewest also, can you try a very very large bolus (> 25.6 u) on an x22 as well, to see if that overflows the bytes? Please disconnect first ;-)

channemann, 2015-10-06:
Fixed by https://github.com/bewest/decoding-carelink/pull/134

closed 2016-05-30 by bewest