openaps/openaps #60: Permissions Issue (Related to Fuser?)

channemann, 2015-10-07:
Periodically seeing some issues like this; can occur at various places in the loop (cgm and pump reports), seemingly at random:

```
cgms://JSON/iter_glucose_hours/recent_glucose.json
reporting recent_glucose.json
Traceback (most recent call last):
  File "/usr/local/bin/openaps-report", line 6, in <module>
    exec(compile(open(__file__).read(), __file__, 'exec'))
  File "/home/pi/openaps/bin/openaps-report", line 82, in <module>
    app( )
  File "/home/pi/openaps/openaps/cli/__init__.py", line 44, in __call__
    self.run(self.args)
  File "/home/pi/openaps/bin/openaps-report", line 75, in run
    output = app(args, self)
  File "/home/pi/openaps/openaps/cli/subcommand.py", line 50, in __call__
    return self.method.main(args, app)
  File "/home/pi/openaps/openaps/reports/invoke.py", line 47, in main
    repo.index.add([report.name])
  File "/usr/local/lib/python2.7/dist-packages/GitPython-1.0.1-py2.7.egg/git/index/base.py", line 726, in add
    entries_added.extend(self._entries_for_paths(paths, path_rewriter, fprogress, entries))
  File "/usr/local/lib/python2.7/dist-packages/GitPython-1.0.1-py2.7.egg/git/util.py", line 54, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/GitPython-1.0.1-py2.7.egg/git/index/util.py", line 84, in set_git_working_dir
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/GitPython-1.0.1-py2.7.egg/git/index/base.py", line 612, in _entries_for_paths
    entries_added.append(self._store_path(filepath, fprogress))
  File "/usr/local/lib/python2.7/dist-packages/GitPython-1.0.1-py2.7.egg/git/index/base.py", line 580, in _store_path
    istream = self.repo.odb.store(IStream(Blob.type, st.st_size, stream))
  File "/usr/local/lib/python2.7/dist-packages/gitdb-0.6.4-py2.7-linux-armv7l.egg/gitdb/db/git.py", line 73, in store
    return self._loose_db.store(istream)
  File "/usr/local/lib/python2.7/dist-packages/gitdb-0.6.4-py2.7-linux-armv7l.egg/gitdb/db/loose.py", line 232, in store
    rename(tmp_path, obj_path)
OSError: [Errno 13] Permission denied
```

bewest, 2015-10-07:
I think this is leftover from setting up as root, or running `sudo git pull` or similar would result in this kind of thing.

bewest, 2016-05-29:
Closing for now, we've been fixing up docs and recommendations in chat channels to avoid using root, and I just tweaked (yet again) some udev rules with should help make that easier.

closed 2016-05-29 by bewest