openaps/openaps #65: openaps use mypump sensitivities vs. sensitivies

tghoward, 2015-10-24:
A bit embarrassed to be pointing this out, especially because a fix could break everyone's loop, but just in case there is interest in fixing this typo -- if it even IS a typo:  the openaps use arguments for pumps has "sensitivies" as an option, which has bitten me a few times because I always want to type "sensitivities".   Perhaps sensitivies could stay in there as deprecated for a little while while folks switch over... again, if that's desired. Tim

oskarpearson, 2015-10-26:
Heya

Should I change this universally? Or only in python files?

There are a lot of occurrences in the analysis directory:

```
analysis/578398/ReadHistoryData-page-10.json:      "insulin_sensitivies": [
analysis/578398/ReadHistoryData-page-10.json:      "insulin_sensitivies": [
analysis/578398/ReadHistoryData-page-10.json:      "insulin_sensitivies": [
analysis/578398/ReadHistoryData-page-10.json:      "insulin_sensitivies": [
analysis/578398/ReadHistoryData-page-10.markdown:             'insulin_sensitivies': [{'_offset': 0,
analysis/578398/ReadHistoryData-page-10.markdown:           'insulin_sensitivies': [{'_offset': 0,
analysis/578398/ReadHistoryData-page-10.markdown:             'insulin_sensitivies': [{'_offset': 0,
analysis/578398/ReadHistoryData-page-10.markdown:           'insulin_sensitivies': [{'_offset': 0,
analysis/model-554/ReadHistoryData-page-0.data.markdown:             'insulin_sensitivies': [{'_offset': 0,
analysis/model-554/ReadHistoryData-page-0.data.markdown:           'insulin_sensitivies': [{'_offset': 0,
analysis/model-554/ReadHistoryData-page-0.data.markdown:             'insulin_sensitivies': [{'_offset': 0,
analysis/model-554/ReadHistoryData-page-0.data.markdown:           'insulin_sensitivies': [{'_offset': 0,
analysis/model-554/ReadHistoryData-page-2.data.markdown:             'insulin_sensitivies': [{'_offset': 0,
analysis/model-554/ReadHistoryData-page-2.data.markdown:           'insulin_sensitivies': [{'_offset': 0,
decocare/history.py:  insulin_sensitivies = tail[0:(num*2)]
decocare/history.py:  , insulin_sensitivies=decode_insulin_sensitivies(insulin_sensitivies)
decocare/history.py:  # , _insulin_sensitivies=str(insulin_sensitivies).encode('hex')
decocare/history.py:  # , is_len=len(insulin_sensitivies)
decocare/history.py:def decode_insulin_sensitivies (data):
decocare/models/__init__.py:  read_insulin_sensitivies = Task(commands.ReadInsulinSensitivities)
history.json:[{"_type": "MResultTotals", "_description": "MResultTotals 2015-06-02T00:00:00 head[5], body[0] op[0x07]", "date": 1433199600000.0, "timestamp": "2015-06-02T00:00:00", "_body": "", "_head": "0700000a5c", "_date": "610f"}, {"_type": "BasalProfileStart", "_de
d[2], body[0] op[0x1e]", "date": 1432049794000.0, "timestamp": "2015-05-19T16:36:34", "_body": "", "_head": "1e65", "_date": "626410130f"}, {"_type": "PumpSuspend", "_description": "PumpSuspend 2015-05-19T16:34:16 head[2], body[0] op[0x1e]", "date": 1432049656000.0, "ti
```

oskarpearson, 2015-10-26:
I'm pretty sure the history.json in the base directory can be removed? It looks like a bad PR/merge.

oskarpearson, 2015-10-26:
https://github.com/bewest/decoding-carelink/commit/3b474bc87076b022d2502603cff5b9ed4c03e4fc fixes part of this at least.

tghoward, 2015-12-06:
interesting -- seems to be fixed for the MM 522, but not for the MM 723? Is that possible?

```
pi@betacb ~/bcb-s-new $ openaps use pump723 
bolus                        iter_pump_hours              read_carb_ratios             read_selected_basal_profile  set_temp_basal
filter_glucose_date          model                        read_clock                   read_settings                settings
filter_isig_date             mytest                       read_current_glucose_pages   read_status                  status
-h                           read_basal_profile_A         read_current_history_pages   read_temp_basal              suspend_pump
--help                       read_basal_profile_B         read_glucose_data            reservoir                    
iter_glucose                 read_basal_profile_std       read_history_data            resume_pump                  
iter_glucose_hours           read_battery_status          read_insulin_sensitivies     scan                         
iter_pump                    read_bg_targets              read_insulin_sensitivities   Session                      
pi@betacb ~/bcb-s-new $ openaps use pump723 read_insulin_sensitivities
Traceback (most recent call last):
  File "/usr/local/bin/openaps-use", line 4, in <module>
    __import__('pkg_resources').run_script('openaps==0.0.8', 'openaps-use')
  File "/usr/local/lib/python2.7/dist-packages/setuptools-18.4-py2.7.egg/pkg_resources/__init__.py", line 735, in run_script
  File "/usr/local/lib/python2.7/dist-packages/setuptools-18.4-py2.7.egg/pkg_resources/__init__.py", line 1652, in run_script
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.8-py2.7.egg/EGG-INFO/scripts/openaps-use", line 63, in <module>
    app( )
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.8-py2.7.egg/openaps/cli/__init__.py", line 44, in __call__
    self.run(self.args)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.8-py2.7.egg/EGG-INFO/scripts/openaps-use", line 57, in run
    output = app(args, self)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.8-py2.7.egg/openaps/uses/__init__.py", line 92, in __call__
    return self.method.selected(args)(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.8-py2.7.egg/openaps/uses/__init__.py", line 31, in __call__
    return self.method(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.8-py2.7.egg/openaps/uses/use.py", line 45, in __call__
    output = self.main(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.8-py2.7.egg/openaps/vendors/medtronic.py", line 213, in main
    return getattr(self.pump.model, name)(**self.get_params(args))
AttributeError: 'Model723' object has no attribute 'read_insulin_sensitivities'
pi@betacb ~/bcb-s-new $ openaps use pump723 read_insulin_sensitivies
{
  "sensitivities": [
    {
      "i": 0, 
      "start": "00:00:00", 
      "sensitivity": 70, 
      "offset": 0, 
      "x": 0
    }
  ], 
  "action_type": "Fast", 
  "first": 1
}pi@betacb ~/bcb-s-new $ 

```

bewest, 2016-05-29:
@tghoward, are you still having an issue with this?

closed 2018-03-09 by tghoward