openaps/openaps #69: Insulin sensitivities issue with mmol/l pumps

oskarpearson, 2015-12-03:
Hi

When running the read_insulin_sensitivies report, the result is as below - with the sensitivity being 2.2 rather than a more normal 22.

I'm capturing this as I'm not convinced that openaps correctly handles the mmol/L units correctly. That'd lead to hypo actions. I may close the ticket, but I don't want to lose track of this.
```
{
  "units": "mmol/L",
  "sensitivities": [
    {
      "i": 0,
      "start": "00:00:00",
      "sensitivity": 2.2,
      "offset": 0,
      "x": 0
    }
  ],
  "first": 2
}
```

scottleibrand, 2015-12-03:
What version are you using? I thought @bewest fixed this. 

Have you seen oref0 use the bad sensitivity? Do we need to set a minimum sensitivity of like 15 and refuse to run if it's lower?

bewest, 2015-12-06:
I'd like to revisit the loops from a high-level perspective and create a clear requirements document/specification of what the "device" should do.  This is a good example of one of the extra things the device needs to do that often gets lost... it needs to ensure all incoming reports have validated/standardized into reports with proper units.

After we get a nice list/specification of these things, it'll be easier to refactor the uses to ensure everything is happening.  Let's leave this open and try to link it in to some kind of "specifications" sprint with other requirements.

In the meantime, I think you need to configure a few extra uses/devices to ensure that the units conversion is happening.  Is the units/converter not working, it looks like it didn't run on this report.

bewest, 2016-05-30:
Any update here? Looking at adding support for a units preference in the new `use pump config --units` use case.

oskarpearson, 2016-05-30:
I certainly think that your work to template-ize things in https://github.com/openaps/docs/issues/132 is a really good approach - it means people are less likely to get into situations where unit conversions matter. Separating the 'raw' from the 'cooked' values is a reasonable way to go, imho.

Currently all the openaps code appears to fail cleanly if it's presented with data in the wrong data form. Having fail-safe overrides wouldn't hurt, though!

closed 2021-03-22 by oskarpearson