openaps/openaps #70: iter_pump error

tghoward, 2015-12-12:
This error just popped up and is repeating itself. I'd be happy to provide any more information as needed. The error doesn't seem to stop use or report commands from completing.
```
pi@betacb ~/bcb-s $ openaps use pump723 iter_pump 1
ERROR month must be in 1..12 0000   0x1e 0x00                                  ..
ERROR month must be in 1..12 0000   0x1e 0x00                                  ..
ERROR month must be in 1..12 0000   0x1e 0x00                                  ..
[
  {
    "_type": "TempBasalDuration", 
    "duration (min)": 30, 
    "_description": "TempBasalDuration 2015-12-11T18:56:23 head[2], body[0] op[0x16]", 
    "timestamp": "2015-12-11T18:56:23", 
    "_body": "", 
    "_head": "1601", 
    "_date": "d738124b0f"
  }, 
  {
    "_type": "TempBasal", 
    "temp": "absolute", 
    "_description": "TempBasal 2015-12-11T18:56:23 head[2], body[1] op[0x33]", 
    "timestamp": "2015-12-11T18:56:23", 
    "_body": "00", 
    "_head": "3346", 
    "rate": 1.75, 
    "_date": "d738124b0f"
  }
]pi@betacb ~/bcb-s $ openaps report invoke monitor/pumphistory.json
pump723://JSON/iter_pump/monitor/pumphistory.json
ERROR month must be in 1..12 0000   0x1e 0x00                                  ..
ERROR month must be in 1..12 0000   0x1e 0x00                                  ..
ERROR month must be in 1..12 0000   0x1e 0x00                                  ..
reporting monitor/pumphistory.json
pi@betacb ~/bcb-s $ 
```

tghoward, 2015-12-12:
seems to consistently be three lines each time

loudnate, 2015-12-12:
Can you pull your carelink CSV (from carelink.minimed.com) and the raw pump data (using decocare)? There's an unexpected record or record format here.

tghoward, 2015-12-12:
Uploaded to carelink.minimed.com (a feat in itself as we've been having problems doing that) and got the csv from the last couple of days. Will see what I can do with decocare now. Might ask for pointers.

tghoward, 2015-12-12:
got a dump using mm-latest.py. :)

tghoward, 2015-12-12:
ok, see the two new files in this branch:
https://github.com/tghoward/decoding-carelink/tree/pumpData
files are:
https://github.com/tghoward/decoding-carelink/blob/pumpData/CareLink-Export-1449885458492.csv
https://github.com/tghoward/decoding-carelink/blob/pumpData/tgh_latestlogs.json

Hope that is helpful in some way. Thanks Nate.   I think these aren't related, but we also just got some very strange bolus records from openaps -- claiming there was a 70 unit bolus, for example. You might see it in these data, but if not I have other exports that will show that if you think they are related and worth looking at at the same time. Tim

tghoward, 2015-12-12:
Because the data are on the same CareLink Export and I'm now seeing that it does look like another mis-read by decocare, I uploaded the screwy bolus records too.   There are three files here:
https://github.com/tghoward/decoding-carelink/tree/pumpData/bolusErrors
Two files have just the json for the offending records, the third has all the recent Bolus records. If you find the timestamp in the Carelink csv file, MM has no bolus recorded.   

If this is worth pursuing, please let me know if there is anything else I can do.  Also happy to be set straight if I'm doing something wrong!

tghoward, 2016-01-08:
In the csv from carelink.minimed.com (linked above), there are entries named in the "Raw Type" column that don't appear at the end of the DecoCare logs. Such as: SensorCal; SensorCalFactor; SensorStatus; SensorSync; SensorPacket; ChangeSensorSetupConfig2; ChangeSensorSetup2. 

The decocare logs do show the problematic error.

tghoward, 2016-01-10:
Added additional data, using 
mm-send-comm.py tweak ReadHistoryData --page 0 --save

Placed these page dumps (pages 0-32) here.
 
https://github.com/tghoward/decoding-carelink/tree/thowdat/logs/2016-01-09