openaps/openaps #76: No LICENSE.txt?

mplewis, 2016-01-06:
Hello! I saw this in the readme:

> By proceeding using these tools or any piece within, you agree to the copyright (see LICENSE.txt for more information) and release any contributors from liability.

However, I can't find LICENSE.txt. Is this in another repo?

stavlor, 2016-01-06:
hmm good catch, check here perhaps https://github.com/openaps/oref0/blob/master/LICENSE.txt

scottleibrand, 2016-01-06:
That's the LICENSE I did for oref0 (the algorithms and higher-level stuff
that might be useful commercially).  I think the underlying openaps toolkit
might have originally been licensed under GPL. Most likely @bewest would
know.

-Scott

bewest, 2016-05-29:
Added LICENSE and applied my copyright here https://github.com/openaps/openaps/blob/master/LICENSE.txt :smiley_cat: Closing.

closed 2016-05-29 by bewest