openaps/openaps #78: Looking for advice

nlewis93, 2016-01-26:
Hi everyone! My name is Nolan. I recently was introduced to the Nightscout community by my family out of interest in the artificial pancreas system. My father is a T1 diabetic who among other things also suffers from celiacs disease. As soon as i read about the program and saw how it could effect his life i was instantly on board. 
However, I dont have a major background in computer science or software engineering. I study Information Science as a junior at the University of Pittsburgh but this isnt entirely helpful for writing the algorithms and in-depth code implementation that this project requires. My experience includes taking two courses each in C++ and java programming and im ok with navigating Linux for the more basic things from working with pi in the past.  My question to all of you is do you think a person of my skill level could safely and effectively venture into this project? 
I'm willing to learn and i realize it might take a large amount of time in doing so but its the least i could do for someone who has done so much for me throughout my life. I do want to be cautious however, i realize fiddling with a medical device is something that deserves the utmost care. 

If you would like to email me personally you can do so at Nlewis5525@gmail.com 
Thanks for any Help! 
-Nolan

scottleibrand, 2016-01-26:
Sounds like you have more than enough technical experience to dive right
in.  We've had folks with absolutely no Linux experience get themselves up
and running successfully, albeit with a bit longer learning curve than
folks more familiar with how to get around a command line.  You really
don't need any particular programming or algorithms background: mostly
you'll just be assembling one-liners that collect and process various data
using tools that have already been created.  Some programming experience is
useful if you want to drill down into the code to see exactly how the
recommendations are generated, but that's not a requirement to successfully
build and safely test a loop.

Take a look at https://github.com/openaps/docs, and then hop into
https://gitter.im/nightscout/intend-to-bolus with any questions.  If you're
committed to a bit of learning, you'll be up and running before you know
it.  :-)

-Scott

nlewis93, 2016-01-27:
Scottleibrand, Hello and thanks for the quick reply! 
I'm glad to hear that others have tried and been successful before me. However, I feel like i'm a bit confused with the systems overall. From what i've learned through a bit of research i have (in a broad scope) two ways to go. One being the use of an android to pull data from the cgm and pump to display onto other devices for a little more safety and overall access and the second being the implementation of a raspberry pi and multiple algorithms to fully automate the pump's processes. In addition to that I assumed I had to write all of those pieces of code to make it work. An example of what i mean is building the manual system in Phase two on the Open Artificial Pancreas System. 
https://openaps.gitbooks.io/building-an-open-artificial-pancreas-system/content/docs/Build-manual-system/considerations.html
Again thanks so much for any help! 
-Nolan

scottleibrand, 2016-01-27:
The former is what we call Nightscout.  It doesn't give you any control over the pump.  The latter is what we call OpenAPS.

Your assumption is no longer correct.  Read the rest of the docs: the oref0 toolset has all the algorithms and code you need.

And definitely hop into Gitter with questions.  There are dozens of people who can answer them.

closed 2016-01-27 by scottleibrand