openaps/openaps #83: Invoking reports throws an error randomly

lukasz-kaniowski, 2016-03-13:
I'm fallowing documentation and often I'm getting same exception for different reports:
i.e: ```$ openaps report invoke settings/insulin_sensitivities.json``` results with:

 
```
pump://JSON/read_insulin_sensitivies/settings/insulin_sensitivities.json
settings/insulin_sensitivities.json  raised  'Pump' object has no attribute 'model'
Traceback (most recent call last):
  File "/usr/local/bin/openaps-report", line 4, in <module>
    __import__('pkg_resources').run_script('openaps==0.0.10', 'openaps-report')
  File "build/bdist.linux-armv7l/egg/pkg_resources/__init__.py", line 726, in run_script
  File "build/bdist.linux-armv7l/egg/pkg_resources/__init__.py", line 1484, in run_script
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/EGG-INFO/scripts/openaps-report", line 82, in <module>
    app( )
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/openaps/cli/__init__.py", line 44, in __call__
    self.run(self.args)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/EGG-INFO/scripts/openaps-report", line 75, in run
    output = app(args, self)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/openaps/cli/subcommand.py", line 52, in __call__
    return self.method.main(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/openaps/reports/invoke.py", line 40, in main
    output = task.method(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/openaps/uses/use.py", line 45, in __call__
    output = self.main(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/openaps/vendors/medtronic.py", line 217, in main
    return getattr(self.pump.model, name)(**self.get_params(args))
AttributeError: 'Pump' object has no attribute 'model'
```

or `$ openaps report invoke settings/bg_targets.json`

```
pump://JSON/read_bg_targets/settings/bg_targets.json
settings/bg_targets.json  raised  'Pump' object has no attribute 'model'
Traceback (most recent call last):
  File "/usr/local/bin/openaps-report", line 4, in <module>
    __import__('pkg_resources').run_script('openaps==0.0.10', 'openaps-report')
  File "build/bdist.linux-armv7l/egg/pkg_resources/__init__.py", line 726, in run_script
  File "build/bdist.linux-armv7l/egg/pkg_resources/__init__.py", line 1484, in run_script
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/EGG-INFO/scripts/openaps-report", line 82, in <module>
    app( )
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/openaps/cli/__init__.py", line 44, in __call__
    self.run(self.args)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/EGG-INFO/scripts/openaps-report", line 75, in run
    output = app(args, self)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/openaps/cli/subcommand.py", line 52, in __call__
    return self.method.main(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/openaps/reports/invoke.py", line 40, in main
    output = task.method(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/openaps/uses/use.py", line 45, in __call__
    output = self.main(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/openaps/vendors/medtronic.py", line 222, in main
    return getattr(self.pump.model, name)(**self.get_params(args))
AttributeError: 'Pump' object has no attribute 'model'
```

If I invoke it few times the report in the end is generated though.

scottleibrand, 2016-03-13:
That usually indicates that radio communications did not complete successfully.  You can try placing the Carelink stick closer to the pump, but be aware that the range is generally terrible.  Once you have a loop up and working with the Carelink, you might want to look into getting a TI stick to use with mmeowlink.

lukasz-kaniowski, 2016-03-13:
What would be tolerable distance that I should use for now? Currently it's about 3 meters.

scottleibrand, 2016-03-13:
Reliable range is more like 1 meter.

closed 2016-03-13 by lukasz-kaniowski