openaps/openaps #86: Timezone doesn't handle daylight savings time well

audiefile, 2016-03-20:
All times and time zones were set correctly but pump history wasn't properly zoned so treatments were displayed in Nightscout as off by one hour (one hour into the future). (Originally posted in oref0 by mistake)

bewest, 2016-05-29:
The new default behavior is to leave the `timezone = ` field blank/empty.  This causes it to use to track with the local system's time, which if using ntp, will traverse across DST events nicely.

If you remove the zoning report, and re-add, it, you can compare with `show --cli` that the `timezone = ` field is now blank, and if the system time updates, the locale's offset is followed as expected.  To get the old behavior (or to force everything to Zulu/UTC) set the timezone to the desired zone, (`GMT` or similar).  HTH.

scottleibrand, 2016-11-07:
This particular aspect of DST support seems to be working well now that almost everyone has blank `timezone =` fields.

closed 2016-11-07 by scottleibrand