openaps/openaps #87: race condition in openaps configuration

scottleibrand, 2016-03-21:
When updating an alias or report using `openaps alias add`, `openaps report add`, or `openaps import`, sometimes the configuration will show updated (it will echo back the new config), but then when you run `openaps (alias|report) show` immediately afterward, the config has reverted to its previous state.  (Usually I don't notice this until some time later, when the config I'm relying on to be updated doesn't seem to be working, for no good reason.)

I suspect this is because another openaps operation is ongoing at the time the (`add`|`import`) was executed, and when that operation completes, it writes its cached copy of the config, overwriting the updated config.

bewest, 2016-03-21:
Should be fixed in https://github.com/openaps/openaps/pull/80/commits/1c7996b73ab288fc79e7cbb27143bde28d3d799a

From the manual, I thought this was interesting:

* http://gitpython.readthedocs.org/en/stable/reference.html#git.index.base.IndexFile.add

> *write_extension_data* – If True, extension data will be written back to the index. This can lead to issues in case it is containing the ‘TREE’ extension, which will cause the git commit command to write an old tree, instead of a new one representing the now changed index. This doesn’t matter if you use IndexFile.commit(), which ignores the TREE extension altogether. You should set it to True if you intend to use IndexFile.commit() exclusively while maintaining support for third-party extensions. Besides that, you can usually safely ignore the built-in extensions when using GitPython on repositories that are not handled manually at all.

bewest, 2016-03-21:
Also, to clarify, root problem is that changes to `openaps.ini` are never being added to index after https://github.com/openaps/openaps/commit/ca313055fed1493c88d52a1b093661c3a1430b1f

commit cdc09cc2 (2016-03-21): properly fix #87, regression from ca313055fed1493c88d52a1b093661c3a1430b1f

* https://github.com/openaps/openaps/commit/ca313055fed1493c88d52a1b093661c3a1430b1f
This emulates what git commit -a does.
closed 2016-04-28 by bewest