openaps/openaps #88: read_basal_profile_std only report the first 20 values

iainct, 2016-04-02:
`openaps report add settings/basal_profile.json JSON pump read_basal_profile_std`
only report the first 20 values

if like me you have a setting for every hour of the day at 8pm that becomes the basal value until midnight.

This coupled with the logic below results in some dangerous conditions around 10pm when openaps bounces between the 8pm basal rate and zero as shown in the below charts.

```
history.js --> function calcTempTreatments (inputs)
var netBasalRate = tempHistory[i].rate-profile_data.current_basal;
var netBasalAmount = Math.round(netBasalRate*tempHistory[i].duration*10/6)/100
```

[image]

scottleibrand, 2016-04-02:
Can you please post the basal_profile.json it wrote out?

iainct, 2016-04-02:
`
[
  {
    "i": 0, 
    "start": "00:00:00", 
    "rate": 0.6000000000000001, 
    "minutes": 0
  }, 
  {
    "i": 1, 
    "start": "01:00:00", 
    "rate": 0.7000000000000001, 
    "minutes": 60
  }, 
  {
    "i": 2, 
    "start": "02:00:00", 
    "rate": 0.75, 
    "minutes": 120
  }, 
  {
    "i": 3, 
    "start": "03:00:00", 
    "rate": 0.7000000000000001, 
    "minutes": 180
  }, 
  {
    "i": 4, 
    "start": "04:00:00", 
    "rate": 0.6000000000000001, 
    "minutes": 240
  }, 
  {
    "i": 5, 
    "start": "05:00:00", 
    "rate": 0.6000000000000001, 
    "minutes": 300
  }, 
  {
    "i": 6, 
    "start": "06:00:00", 
    "rate": 0.7000000000000001, 
    "minutes": 360
  }, 
  {
    "i": 7, 
    "start": "07:00:00", 
    "rate": 0.45, 
    "minutes": 420
  }, 
  {
    "i": 8, 
    "start": "08:00:00", 
    "rate": 0.4, 
    "minutes": 480
  }, 
  {
    "i": 9, 
    "start": "09:00:00", 
    "rate": 0.4, 
    "minutes": 540
  }, 
  {
    "i": 10, 
    "start": "10:00:00", 
    "rate": 0.35000000000000003, 
    "minutes": 600
  }, 
  {
    "i": 11, 
    "start": "11:00:00", 
    "rate": 0.35000000000000003, 
    "minutes": 660
  }, 
  {
    "i": 12, 
    "start": "12:00:00", 
    "rate": 0.35000000000000003, 
    "minutes": 720
  }, 
  {
    "i": 13, 
    "start": "13:00:00", 
    "rate": 0.35000000000000003, 
    "minutes": 780
  }, 
  {
    "i": 14, 
    "start": "14:00:00", 
    "rate": 0.35000000000000003, 
    "minutes": 840
  }, 
  {
    "i": 15, 
    "start": "15:00:00", 
    "rate": 0.25, 
    "minutes": 900
  }, 
  {
    "i": 16, 
    "start": "16:00:00", 
    "rate": 0.2, 
    "minutes": 960
  }, 
  {
    "i": 17, 
    "start": "17:00:00", 
    "rate": 0.2, 
    "minutes": 1020
  }, 
  {
    "i": 18, 
    "start": "18:00:00", 
    "rate": 0.8, 
    "minutes": 1080
  }, 
  {
    "i": 19, 
    "start": "19:00:00", 
    "rate": 1.4000000000000001, 
    "minutes": 1140
  }, 
  {
    "i": 20, 
    "start": "20:00:00", 
    "rate": 1.5, 
    "minutes": 1200
  }
]
`

iainct, 2016-04-02:
I had basal rate values for 9pm, 10pm, and 11pm, but these were not used putting the implementation into a dangerous state that i only found when i went live on an overnight test...

scottleibrand, 2016-04-02:
Do your basals drop dramatically after 8pm? Was it giving the 8pm basal until it saw BG start to fall?

bewest, 2016-04-02:
How many rates do you have in the pump?

Also, can you run something like this:
(`mmeowlink-send.py`works same way as `mm-send-comm.py`)
```
$ mmeowlink-send.py --serial 571347 tweak ReadProfile_STD512 
subg_rfspy status: OK
Version: subg_rfspy 0.6
```
GOT MODEL ReadPumpModel:size[64]:data:'522'
<class 'decocare.commands.ReadProfile_STD512'> {}
response: ReadProfile_STD512:size[64]:data:[{'i': 0, 'start': '00:00:00', 'rate': 0.7000000000000001, 'minutes': 0}, {'i': 1, 'start': '06:30:00', 'rate': 1.05, 'minutes': 390}, {'i': 2, 'start': '09:30:00', 'rate': 1.1, 'minutes': 570}, {'i': 3, 'start': '13:30:00', 'rate': 0.9, 'minutes': 810}]
hexdump:
```python
0000   0x1c 0x00 0x00 0x2a 0x00 0x0d 0x2c 0x00    ...*..,.
0008   0x13 0x24 0x00 0x1b 0x00 0x00 0x00 0x00    .$......
0010   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0018   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0020   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0028   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0030   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0038   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
```
#### decoded:
```python
[{'i': 0, 'start': '00:00:00', 'rate': 0.7000000000000001, 'minutes': 0}, {'i': 1, 'start': '06:30:00', 'rate': 1.05, 'minutes': 390}, {'i': 2, 'start': '09:30:00', 'rate': 1.1, 'minutes': 570}, {'i': 3, 'start': '13:30:00', 'rate': 0.9, 'minutes': 810}]
```

Something like this to get the raw data, along with knowing what the expected answer is will enable us to fix the issue.  It would also be nice to consider calling Medtronic and asking them for more details on this issue.  If they could provide me with the manual for these details, I can ensure that the details here matches.

iainct, 2016-04-03:
@bewest  I had 24 values for basal profile, 1 for every our of the day, but it seams as though the search only went from `i = 1..20` vs `i = 1..end` unfortunately i am not an official Medtronic customer so I do not have grounds to start a dialogue, I'm sorry about that as i understand the more queries they get the higher the status of this lack of user functionality issued. I have dropped the values right down now to prevent this bug.

I tired to run `mmeowlink-send.py` but encountered the following error.

`pi@raspberrypi:~/mmeowlink-source/bin $ sudo python mmeowlink-send.py --serial ###### tweak ReadProfile_STD512 
Traceback (most recent call last):
  File "mmeowlink-send.py", line 7, in <module>
    app.run(None)
  File "/home/pi/src/decoding-carelink/decocare/helpers/cli.py", line 113, in run
    self.prelude(args)
  File "/home/pi/mmeowlink-source/mmeowlink/cli/messages.py", line 15, in prelude
    self.link = link = LinkBuilder().build(args.radio_type, args.port)
  File "/home/pi/mmeowlink-source/mmeowlink/link_builder.py", line 13, in build
    raise UnknownLinkType("Unknown radio type '%s' - check parameters" % radio_type)
mmeowlink.exceptions.UnknownLinkType: Unknown radio type 'None' - check parameters
`

further to this and its probably an issue by itself; running oref0-calculate-iob.js with the attached inputs gives a dangerous output:

[calculate-iob-inputs.zip](https://github.com/openaps/openaps/files/201145/calculate-iob-inputs.zip)

`
{
  "netbasalinsulin": 0.9, 
  "activity": 0.0014, 
  "basaliob": -0.305, 
  "hightempinsulin": 1.5, 
  "bolussnooze": 0, 
  "iob": -0.305

}
`

iainct, 2016-04-03:
@bewest to be clear  by dangerous I was referring to the negative iob, which at low blood sugar causes the eventual blood sugar to be much higher than it ever will be. This happened whilst I was hypo and the temp-basal was increased!

cheers

scottleibrand, 2016-04-03:
Do you have a copy of the currenttemp.json and glucose.json at the time that it high-temped?

iainct, 2016-04-03:
@scottleibrand here you go, and thanks for the tip on autosens, every time I try working through that i've corrupted my repo, that is a job for one night later this week.
cheers,
Iain

[glucose_currenttemp.zip](https://github.com/openaps/openaps/files/201153/glucose_currenttemp.zip)

scottleibrand, 2016-04-03:
Running determine-basal (from dev) on those inputs gives me:

```
$ oref0-determine-basal iob.json temp_basal.json glucose.json profile.json
{"delta":4,"glucose":81,"avgdelta":1}
{"duration":24,"rate":0.575,"temp":"absolute"}
{"iob":-1.488,"activity":-0.0108,"bolussnooze":0,"basaliob":-1.488,"netbasalinsulin":-1.45,"hightempinsulin":0.5}
{"max_iob":2,"type":"current","dia":3,"current_basal":1.4,"max_daily_basal":1.5,"max_basal":2,"min_bg":106,"max_bg":145,"sens":40}
{"temp":"absolute","bg":81,"tick":"+4","eventualBG":134,"snoozeBG":134,"mealAssist":"Off: Carbs: undefined Boluses: undefined Target: 126 Deviation: -7 BGI: 2.16","reason":"Eventual BG 134>106 but Avg. Delta 1.00 < Exp. Delta 1.9; setting current basal of 1.4 as temp","duration":30,"rate":1.4}
```

So it looks like it's doing the right thing as far as setting the current basal as temp, but obviously if the current basal is actually much lower than what openaps is reporting, that could be too high.

If you drop the 7-9pm basals down to something more like the 6-7pm basal, that should be a lot safer.  Obviously you'll also need to bolus a bit more for dinner if you do that, or set up meal-assist so oref0 can do so.

bewest, 2016-04-03:
There's an additional switch for `mmeowlink-send.py --radio_type subgrfspy ...` keep in mind, for your first run you might need to add `--init` as well.

bewest, 2016-04-03:
See: https://github.com/bewest/decoding-carelink/pull/175/commits/c90fe6773c4f32ebea0f1a2c18bd0d4e16eaa0b0

`$ mm-send-comm.py  tweak ReadProfile_A512`
## do stuff with an insulin pump over RF
using ` Namespace(autoinit=False, begin=None, bytesPerRecord=None, command='tweak', descr=None, dryrun=False, effectTime=None, end=None, init=False, maxRecords=None, name=None, no_postlude=False, no_prelude=False, no_rf_prelude=False, other='ReadProfile_A512', page=None, params=None, port='scan', postfix=None, prefix=None, prefix_path='', save=False, saveall=False, serial='665455', session_life=10, verbose=None) `
```
```
```
```
```
```
```
```
```javascript
{'radio': {'errors.crc': 0,
           'errors.naks': 0,
           'errors.sequence': 0,
           'errors.timeouts': 0,
           'packets.received': 11L,
           'packets.transmit': 12L},
 'usb': {'errors.crc': 0,
         'errors.naks': 0,
         'errors.sequence': 0,
         'errors.timeouts': 0,
         'packets.received': 55L,
         'packets.transmit': 55L}}
```
```
```
### PUMP MODEL: `ReadPumpModel:size[64]:data:'522'`
<class 'decocare.commands.ReadProfile_A512'> {}
response: ReadProfile_A512:size[128]:data:[{'i': 0, 'start': '00:00:00', 'rate': 0.1, 'minutes': 0}, {'i': 1, 'start': '01:00:00', 'rate': 0.1, 'minutes': 60}, {'i': 2, 'start': '02:00:00', 'rate': 0.15000000000000002, 'minutes': 120}, {'i': 3, 'start': '03:00:00', 'rate': 0.15000000000000002, 'minutes': 180}, {'i': 4, 'start': '04:00:00', 'rate': 0.25, 'minutes': 240}, {'i': 5, 'start': '05:00:00', 'rate': 0.25, 'minutes': 300}, {'i': 6, 'start': '06:00:00', 'rate': 0.30000000000000004, 'minutes': 360}, {'i': 7, 'start': '07:00:00', 'rate': 0.35000000000000003, 'minutes': 420}, {'i': 8, 'start': '08:00:00', 'rate': 0.4, 'minutes': 480}, {'i': 9, 'start': '09:00:00', 'rate': 0.4, 'minutes': 540}, {'i': 10, 'start': '09:30:00', 'rate': 0.05, 'minutes': 570}, {'i': 11, 'start': '10:00:00', 'rate': 0.05, 'minutes': 600}, {'i': 12, 'start': '10:30:00', 'rate': 0.1, 'minutes': 630}, {'i': 13, 'start': '11:00:00', 'rate': 0.1, 'minutes': 660}, {'i': 14, 'start': '11:30:00', 'rate': 0.15000000000000002, 'minutes': 690}, {'i': 15, 'start': '12:00:00', 'rate': 0.25, 'minutes': 720}, {'i': 16, 'start': '12:30:00', 'rate': 0.05, 'minutes': 750}, {'i': 17, 'start': '13:00:00', 'rate': 0.1, 'minutes': 780}, {'i': 18, 'start': '13:30:00', 'rate': 0.15000000000000002, 'minutes': 810}, {'i': 19, 'start': '14:00:00', 'rate': 0.05, 'minutes': 840}, {'i': 20, 'start': '14:30:00', 'rate': 0.1, 'minutes': 870}, {'i': 21, 'start': '15:00:00', 'rate': 0.15000000000000002, 'minutes': 900}, {'i': 22, 'start': '15:30:00', 'rate': 0.05, 'minutes': 930}, {'i': 23, 'start': '16:00:00', 'rate': 0.2, 'minutes': 960}, {'i': 24, 'start': '16:30:00', 'rate': 0.05, 'minutes': 990}, {'i': 25, 'start': '17:00:00', 'rate': 0.1, 'minutes': 1020}, {'i': 26, 'start': '17:30:00', 'rate': 0.15000000000000002, 'minutes': 1050}, {'i': 27, 'start': '18:00:00', 'rate': 0.25, 'minutes': 1080}, {'i': 28, 'start': '18:30:00', 'rate': 0.05, 'minutes': 1110}, {'i': 29, 'start': '19:00:00', 'rate': 0.1, 'minutes': 1140}, {'i': 30, 'start': '19:30:00', 'rate': 0.2, 'minutes': 1170}, {'i': 31, 'start': '20:00:00', 'rate': 0.05, 'minutes': 1200}, {'i': 32, 'start': '20:30:00', 'rate': 0.1, 'minutes': 1230}, {'i': 33, 'start': '21:00:00', 'rate': 0.15000000000000002, 'minutes': 1260}, {'i': 34, 'start': '21:30:00', 'rate': 0.05, 'minutes': 1290}, {'i': 35, 'start': '22:00:00', 'rate': 0.1, 'minutes': 1320}, {'i': 36, 'start': '22:30:00', 'rate': 0.15000000000000002, 'minutes': 1350}, {'i': 37, 'start': '23:00:00', 'rate': 0.05, 'minutes': 1380}, {'i': 38, 'start': '23:30:00', 'rate': 0.1, 'minutes': 1410}]
hexdump:
```python
0000   0x04 0x00 0x00 0x04 0x00 0x02 0x06 0x00    ........
0008   0x04 0x06 0x00 0x06 0x0a 0x00 0x08 0x0a    ........
0010   0x00 0x0a 0x0c 0x00 0x0c 0x0e 0x00 0x0e    ........
0018   0x10 0x00 0x10 0x10 0x00 0x12 0x02 0x00    ........
0020   0x13 0x02 0x00 0x14 0x04 0x00 0x15 0x04    ........
0028   0x00 0x16 0x06 0x00 0x17 0x0a 0x00 0x18    ........
0030   0x02 0x00 0x19 0x04 0x00 0x1a 0x06 0x00    ........
0038   0x1b 0x02 0x00 0x1c 0x04 0x00 0x1d 0x06    ........
0040   0x00 0x1e 0x02 0x00 0x1f 0x08 0x00 0x20    .......
0048   0x02 0x00 0x21 0x04 0x00 0x22 0x06 0x00    ..!.."..
0050   0x23 0x0a 0x00 0x24 0x02 0x00 0x25 0x04    #..$..%.
0058   0x00 0x26 0x08 0x00 0x27 0x02 0x00 0x28    .&..'..(
0060   0x04 0x00 0x29 0x06 0x00 0x2a 0x02 0x00    ..)..*..
0068   0x2b 0x04 0x00 0x2c 0x06 0x00 0x2d 0x02    +..,..-.
0070   0x00 0x2e 0x04 0x00 0x2f 0x00 0x00 0x00    ..../...
0078   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0080   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0088   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0090   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
0098   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
00A0   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
00A8   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
00B0   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
00B8   0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00    ........
```
#### decoded:
```python
[{'i': 0, 'start': '00:00:00', 'rate': 0.1, 'minutes': 0}, {'i': 1, 'start': '01:00:00', 'rate': 0.1, 'minutes': 60}, {'i': 2, 'start': '02:00:00', 'rate': 0.15000000000000002, 'minutes': 120}, {'i': 3, 'start': '03:00:00', 'rate': 0.15000000000000002, 'minutes': 180}, {'i': 4, 'start': '04: [... cut ...] nutes': 1380}, {'i': 38, 'start': '23:30:00', 'rate': 0.1, 'minutes': 1410}]
```
### end stats
```
```
```javascript
{'radio': {'errors.crc': 0,
           'errors.naks': 0,
           'errors.sequence': 0,
           'errors.timeouts': 0,
           'packets.received': 15L,
           'packets.transmit': 17L},
 'usb': {'errors.crc': 0,
         'errors.naks': 0,
         'errors.sequence': 0,
         'errors.timeouts': 0,
         'packets.received': 63L,
         'packets.transmit': 63L}}
```

closed 2016-04-28 by bewest