openaps/openaps #91: Timezone vendor not installing correctly

sjolundjohn, 2016-05-03:
Dump from when I try to install to updated core:

pi@raspberrypi:~/test_openaps/tutorial $ openaps vendor add openapscontrib.timezones
No module named recurrent
openapscontrib.timezones doesn't seem to be an importable python module
If it is a python module, try using --path to influence
PYTHONPATH

djjdub, 2016-07-01:
Any luck with this? I'm running into the same issue - slightly different output - but same end results:

    pi@raspberrypi:~/Documents/tutorial-hello $ openaps-vendor add openapscontrib.timezone
    No module named openapscontrib.timezone
    openapscontrib.timezone doesn't seem to be an importable python module
    If it is a python module, try using --path to influence
    PYTHONPATH

I'm not familiar enough with openaps yet to really know how to troubleshoot, so I'm trying to ensure that I don't skip steps to avoid them coming back to haunt me later!

T0-9, 2024-09-08:
@djjdub  Page is gone, you still have a list somewhere?