openaps/openaps #100: auto-sens.json report should sort the data prior to processing

Spazholio, 2016-06-16:
I noticed when I had two different data sources (CGM and NS) that were (apprently) appended one onto the other, the auto-sens.json report would run through one and then the other, resulting in what could possibly be erroneous data.  I'd suggest adding a sort feature right before the report kicks off to ensure that the data is chronologically in order.

scottleibrand, 2016-06-16:
Since you have the offending data handy, could you try adding a sort to the oref0 code and see if you can get it working properly?

Spazholio, 2016-06-17:
Ok, I've got a quick 'n dirty function in place in `oref0/lib/determine-basal/cob-autosens.js`.  Is that the right place to implement it?  Right inside the `detectSensitivityandCarbAbsorption()` function, I'm just sorting `glucose_data` before it's ever acted on.  If this isn't the right place to put it, just let me know and I can move it somewhere more appropriate, then submit a PR.

However, I have a question to either @scottleibrand or @bewest about the code.  And this is just probably a failing of my understanding of the framework.  That's a function called `detectSensitivityandCarbAbsorption()`, right?  Well, I grepped the entire codebase, and I never see that function actually being actually *called* anywhere.  Can you shed some light on this?  This is my first foray into node (but not my first foray into JS).

scottleibrand, 2016-06-27:
the detectSensitivityandCarbAbsorption() function is the default function for that module (module.exports = detectSensitivityandCarbAbsorption;).  so when require('oref0/lib/determine-basal/cob-autosens'), that is the function that gets called.
```
bin/oref0-detect-sensitivity.js:var detect = require('oref0/lib/determine-basal/cob-autosens');
lib/meal/total.js:var calcMealCOB = require('oref0/lib/determine-basal/cob-autosens');
```

I have no strong opinions on where the sorting should happen.  Find a logical spot and make sure it doesn't break anything.  :-)