openaps/openaps #101: Report nightscout/recent-missing-entries.json fails with AttributeError

bfaloona, 2016-06-18:
This gist shows the details of the report, and the Traceback.

https://gist.github.com/bfaloona/6e2e9ea3ee734614631d8f99f0f7d528

bfaloona, 2016-06-18:
I've tried changing the recent-missing-entries.json `reporter` to 'text', and I've tried changing glucose-raw.json `remainder` from entries.json to entries/svg.json

tmecklem, 2016-09-30:
I encountered the same thing recently with elite data. It seems like the output from mm-format-ns-glucose wasn't compatible with the expected input of format-recent-type. I made a change equivalent to this commit and it all started working: https://github.com/tmecklem/oref0/commit/bed40e521191903e03914249250064c306934d19

Is this a version mismatch or something?

mwilliams, 2016-10-03:
> started working: tmecklem/oref0@bed40e5

Is there any reason the above commit shouldn't move upstream?  Still no
luck getting data into Nightscout here.  I have basal histories uploading
just fine, but no glucose history.  I'm sure I'm overlooking something
trivial, and I've done so many revisions on it, my reports and alias's are
a bit messy.  I should take a few steps back and give it a whirl.  Seems
like separate tutorials for Dexom vs Enlite users might be valuable (or
included a built in `generate-nightscout-upload-alias` util that simply
does it all with all the reporting assumptions in place (not sure if that's
pushing things too far).