openaps/openaps #102: Erroneous --glucose "None" --sensor "None" defaults to oref0_glucose report

scottleibrand, 2016-06-28:
added cgm://JSON/oref0_glucose/raw-cgm/raw-entries.json
pi@rpi3 ~/openaps-mmeow $ openaps report show raw-cgm/raw-entries.json --cli
openaps use cgm oref0_glucose --threshold "100" --glucose "None" --sensor "None" --hours "24.0"
pi@rpi3 ~/openaps-mmeow $ openaps report invoke raw-cgm/raw-entries.json
cgm://JSON/oref0_glucose/raw-cgm/raw-entries.json
raw-cgm/raw-entries.json  raised  can't open 'None': [Errno 2] No such file or directory: 'None'
Traceback (most recent call last):
  File "/usr/local/bin/openaps-report", line 6, in <module>
    exec(compile(open(__file__).read(), __file__, 'exec'))
  File "/home/pi/src/openaps/bin/openaps-report", line 82, in <module>
    app( )
  File "/home/pi/src/openaps/openaps/cli/__init__.py", line 51, in __call__
    self.run(self.args)
  File "/home/pi/src/openaps/bin/openaps-report", line 75, in run
    output = app(args, self)
  File "/home/pi/src/openaps/openaps/cli/subcommand.py", line 52, in __call__
    return self.method.main(args, app)
  File "/home/pi/src/openaps/openaps/reports/invoke.py", line 40, in main
    output = task.method(args, app)
  File "/home/pi/src/openaps/openaps/uses/use.py", line 45, in __call__
    output = self.main(args, app)
  File "/home/pi/src/openaps/openaps/vendors/dexcom.py", line 535, in main
    iter_glucose = self.get_glucose_data(params, args)
  File "/home/pi/src/openaps/openaps/vendors/dexcom.py", line 510, in get_glucose_data
    for item in json.load(argparse.FileType('r')(args.glucose)):
  File "/usr/lib/python2.7/argparse.py", line 1142, in __call__
    raise ArgumentTypeError(message % (string, e))
argparse.ArgumentTypeError: can't open 'None': [Errno 2] No such file or directory: 'None'

scottleibrand, 2016-06-28:
Workaround:

pi@rpi3 ~/openaps-mmeow $ openaps report add raw-cgm/raw-entries.json JSON cgm oref0_glucose --hours 24 --glucose "" --sensor ""
added cgm://JSON/oref0_glucose/raw-cgm/raw-entries.json
pi@rpi3 ~/openaps-mmeow $ openaps report show raw-cgm/raw-entries.json --cli
openaps use cgm oref0_glucose --threshold "100" --hours "24.0"
pi@rpi3 ~/openaps-mmeow $ openaps report invoke raw-cgm/raw-entries.json
cgm://JSON/oref0_glucose/raw-cgm/raw-entries.json
reporting raw-cgm/raw-entries.json

bewest, 2016-06-28:
What are you trying to do?  Shouldn't need `--glucose` or `--sensor` unless you want want to read data from a file: `--glucose` will take a file with output from `iter_glucose` and `--sensor` will take a file with output from `iter_sensor`.  To avoid using raw at all, try `--no-raw`.  `--threshold` is used to merge egv and sensor records into one record if they are within the threshold, which probably doesn't need changing.

I'd try: `openaps use cgm oref0_glucose  --hours "24.0"` as documented, suggested unless there's a problem.

"None" isn't a valid file, so it's throwing an error appropriately.  If you use `--glucose` or `--sensor` switches, they must resolve to a file.  Not sure what version you are using, believe several cosmetic issues showing the command have been fixed:

```
 openaps report show raw-cgm/glucose-raw.json 
cgm://JSON/oref0_glucose/raw-cgm/glucose-raw.json

```

scottleibrand, 2016-06-28:
Ooops, looks like the first line didn't paste properly, so you can't see how openaps added the "None" stuff, not me.  Edited the issue above to include that, and here as well:

```
pi@rpi3 ~/openaps-mmeow $ openaps report add raw-cgm/raw-entries.json JSON cgm oref0_glucose --hours 24
added cgm://JSON/oref0_glucose/raw-cgm/raw-entries.json
pi@rpi3 ~/openaps-mmeow $ openaps report show raw-cgm/raw-entries.json --cli
openaps use cgm oref0_glucose --threshold "100" --glucose "None" --sensor "None" --hours "24.0"
```

bewest, 2016-06-28:
Nice, I think I picked up on that towards the end, thanks for clarifying.  Should be fixed in dev (think the bug was introduced in dev as well).

scottleibrand, 2016-06-28:
Confirmed fixed in dev.

closed 2016-06-28 by scottleibrand