openaps/openaps #108: openaps mmtune doesn't read radio settings from pump.ini

sulkaharo, 2016-09-09:
Invoking openaps mmtune seems to always forcefully default to US radio settings regardless of how the pump has been configured in pump.ini, so EU (radio mode WW) users who want to tune need to create a separate shell script to run the tuning

scottleibrand, 2016-09-09:
Perhaps @ps2 or @oskarpearson would know what needs to be fixed here?

ps2, 2016-09-09:
I think this goes beyond mmtune; there are a set of radio parameters that seem to help with WW pumps: https://github.com/oskarpearson/mmeowlink/wiki/Non-USA-pump-settings.  I just went through adding underlying support for radio regions to RileyLinkKit, and support for configuring that in the RileyLink app (and will be doing that soon for Loop).  I think a similar thing needs to be done here. 

There is a 'locale' setting in the MMTune class, but the higher layers need to be modified to read in from pump.ini and pass it along.  Also, there is no support yet for the per-region radio settings.

sulkaharo, 2017-03-26:
Wasn't this resolved by @oskarpearson ?

PieterGit, 2017-03-26:
i integrated @oskarpearson scripts in the devww branch in openaps, see https://github.com/openaps/oref0/pull/357
that pull request is (almost) ready to go to dev. 
this will work for us and ww pumps