openaps/openaps #114: oref0-setup: Using the ShareBLE CGM mode with a G4 receiver doesn't work

jaylagorio, 2017-01-01:
I've been able to get oref0-setup to work on an Edison using a G4 receiver when the receiver is connected via USB OTG but have not been able to make it work using the ShareBLE option. The Bluetooth stack gets installed and appears to be working because Bluetooth tethering to a phone works properly. After that even simple `openaps` commands fail, all the same way:

```
root@tinyaps:~/openaps# openaps use cgm ReadSystemTime
Traceback (most recent call last):
  File "/usr/local/bin/openaps-use", line 63, in <module>
    app( )
  File "/usr/local/lib/python2.7/dist-packages/openaps/cli/__init__.py", line 51, in __call__
    self.run(self.args)
  File "/usr/local/bin/openaps-use", line 57, in run
    output = app(args, self)
  File "/usr/local/lib/python2.7/dist-packages/openaps/uses/__init__.py", line 92, in __call__
    return self.method.selected(args)(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps/uses/__init__.py", line 31, in __call__
    return self.method(args, app)
  File "/root/src/openxshareble/openxshareble/__init__.py", line 97, in __call__
    res = self.ble.run_mainloop_with(run, quit_with_loop=False)
  File "/root/src/Adafruit_Python_BluefruitLE/Adafruit_BluefruitLE/bluez_dbus/provider.py", line 121, in _user_thread_main
    self._return_code = target()
  File "/root/src/openxshareble/openxshareble/__init__.py", line 93, in run
    output = self.main(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps/vendors/dexcom.py", line 181, in main
    result = getattr(self.dexcom, name)(**self.get_params(args))
AttributeError: 'NoneType' object has no attribute 'ReadSystemTime'
```

jaylagorio, 2017-01-15:
In /root/src/openxshare/openxshare/ble/uart.py, in `__init__()`, execution doesn't get past `self._uart = device.find_service(self.UART_SERVICE_UUID)`.

PieterGit, 2017-01-15:
@jaylagorio : it would be nice if this will be fixed and that openxshareble does not depend on bluetooth. Have you used the EXTRA option `dexusb` on `oref0-setup.sh`? That will install my python script which works with USB OTG. Unfortunately I get a lot of problems with the USB device on my explorer board with a European Dexcom share, so I'm still stuck on a raspberry pi 3 rig.

I think this is is a openxshareble issue and should be reported there. There are other known issues, see https://github.com/openaps/openxshareble/issues/5

jaylagorio, 2017-01-16:
@PieterGit I haven't seen anything about EXTRA or what dexusb is. I'm using a US device and using USB OTG haven't found any problems. I didn't realize shareble had a different repo, thanks for pointing me that way. I think that issue is the same as or related to this issue. I'll close this issue and point to it in a comment over there.

closed 2017-01-16 by jaylagorio