openaps/openaps #115: Bolus command always says "recieved":false

scottleibrand, 2017-01-19:
Even when boluses are successful, openaps always returns "recieved":false in the output json.  It usually looks something like:

{"status":"normal","requested":{"units":0.1,"params":[1]},"bolusing":true,"timestamp":"2017-01-18T17:43:10.885651","units":0.1,"suspended":false,"recieved":false,"_type":"BolusRequest"}

cjo20, 2017-01-19:
What's the output of the `mm-bolus.py` command?

scottleibrand, 2017-01-20:
It's failing because I'm not using a Carelink stick.  Do you know how to make it use mmeowlink?  What does mm-bolus.py do that is different from `openaps use pump bolus`?  FWIW, the boluses are working fine, they're just reporting "recieved":false in the output.

cjo20, 2017-01-20:
I think `mmeowlink-bolus` works. I was hoping it would dump the hex data it receives so I could see what it actually gets back from the pump. It doesn't, so I manually added a print in `decocare/commands.py` just before the of statement that sets `received=False`. It expects the first entry in the array to be `0x0c`, but it looks like it is full of `0x0`. Possibly a change in protocol between pump versions?

cjo20, 2017-01-20:
FYI: https://github.com/openaps/decocare/issues/4
https://github.com/openaps/decocare/pull/14/files