openaps/openaps #118: Configuration option to allow clean disabling of git functionality

sulkaharo, 2017-03-26:
AFAIK most users never use the git data for anything and thus the only effect of the git feature is increased energy consumption on the rig, small increase in the probability of disk corruption and the need to run the corruption fix scripts. Thus it'd be logical to allow users to disable the git commits altogether.

scottleibrand, 2017-03-26:
Agreed.  Are you volunteering to work on that?  😃

scottleibrand, 2017-07-11:
https://github.com/openaps/openaps/compare/master...sulkaharo:wip/nogit?expand=1

scottleibrand, 2017-07-15:
Merged this into the https://github.com/openaps/openaps/tree/nogit branch for testing.

scottleibrand, 2017-07-15:
Looks like this isn't backwards compatible?  I upgraded a rig and haven't re-initialized yet, and get this error preventing all openaps commands from working:

ConfigParser.NoOptionError: No option 'git' in section: 'DEFAULT'

scottleibrand, 2017-07-15:
Is the fix for that as simple as a null check at https://github.com/openaps/openaps/blob/nogit/openaps/reports/invoke.py#L50 ?

sulkaharo, 2017-07-15:
Implementing the backward compatibility, sec

scottleibrand, 2017-07-15:
Looks like we also need to check for the existence of args.args[1] at 
https://github.com/openaps/openaps/blob/nogit/openaps/init.py#L6 in order to support running it without the --nogit flag.

bewest, 2017-07-15:
Recommend reconsidering... using git has in fact allowed us to audit and
investigate bugs.

scottleibrand, 2017-07-15:
Yes, git has definitely been useful, especially in the early years when we were troubleshooting lots of bugs in the openaps toolkit and with peoples' custom aliases and reports.  Since we've standardized everyone's `openaps` setups with oref0-setup, though, we've found that the need for json-file auditability has greatly diminished, to the point where I think it's been almost a year since we relied on the `openaps` git history for troubleshooting an issue.  Instead, we either run `openaps` commands manually to reproduce an issue, or look at the pump-loop.log or other log output to see what was happening if there was an issue that is no longer occurring.  This approach is serving us well enough that I don't think git is really needed any longer for most OpenAPS users.

In addition, we routinely (more than once a week) have someone complaining on Gitter or (usually) Facebook that their rig has stopped working due to some sort of git error.  We've automated fixes for most of these (up to and including automatically deleting the ~/myopenaps/.git repo), but would prefer to eliminate the problem entirely for average users by having oref0-setup configure their myopenaps directory with --nogit.  Anyone setting up `openaps` manually can still instantiate their aps directory without the flag, of course, and will still get all the benefits of git auditability.

scottleibrand, 2017-07-16:
Opened PR: https://github.com/openaps/openaps/pull/130