openaps/openaps #121: Wikipedia page

MatthewVita, 2017-04-30:
Dear Maintainers,

My name is Matthew Vita, a programmer in the healthcare industry. I am trying to help build up the high impact "List of open-source health software" Wikipedia page (https://en.wikipedia.org/wiki/List_of_open-source_health_software) and would love to add your software to the list. However, it appears that your project does not have a page on Wikipedia so I ask you to consider the following:

1) Adding a nice article for your project (see the GNU Health article (https://en.wikipedia.org/wiki/GNU_Health) for a really good example and Ginkgo CADx (https://en.wikipedia.org/wiki/Ginkgo_CADx) for a lightweight, but acceptable example).
2) Adding said article to "List of open-source health software" Wikipedia page (https://en.wikipedia.org/wiki/List_of_open-source_health_software).

(If you’re already on Wikipedia, my search-fu wasn’t working when I looked you up so please consider doing only step 2 above)

Thanks,
Matthew

(Personal note: I am very inspired by your project)

MatthewVita, 2017-05-02:
Dear Maintainers,

I just wanted to follow up with a few words in regards to Conflict of Interest (COI). As you can imagine, I have been reaching out to all kinds of mature, reputable, and active open source healthcare projects to help build up this Wikipedia list. However, some project members have (rightly) reached out to me with concerns about COI, so I wanted to clear this up as best as I can. I should first note that my role in this process is of a neutral party that is solely acting in the interests of Wikipedia readers that wish to learn about open source healthcare solutions.

Wikipedia defines COI as "contributing to Wikipedia about yourself, family, friends, clients, employers, or your financial and other relationships. Any external relationship can trigger a conflict of interest." They also go on to say that organizations writing about themselves is a COI.

If you meet one or more of those criteria, I suggest reaching out to a neutral party in your project's community to put together the article. If you do not meet any of those criteria, I suggest putting together an article with material that places the interest of the encyclopedia and its reader above all else.

Assuming you don't have a COI, I encourage you to do as other open source healthcare projects have done by creating a neutral article about the project with the help of the community (users, developers, and subject matter experts) as well as reliable references.

For whatever it's worth, my personal opinion is that if you don't have a COI, it is rather easy to write up an article talking about the software's features, history, usage, and milestones all while being neutral.

Thanks,
Matthew Vita

alistairmcmillan, 2017-07-23:
@MatthewVita Someone created a page: https://en.wikipedia.org/wiki/OpenAPS

MatthewVita, 2017-07-24:
@alistairmcmillan Very cool. Quick question: It says the tech is JavaScript. It's Python, right?

scottleibrand, 2017-07-24:
oref0 (the high-level algorithms) is Javascript, and MIT licensed. The openaps toolkit and decocare are Python. 

We should probably change the license to MIT to reflect the oref0 and openaps licenses, even though they use some GPL components as well.

MatthewVita, 2017-07-25:
@scottleibrand Cool. I just updated the Wiki with this information.

BTW, I am waiting for a response on this https://en.wikipedia.org/wiki/Talk:List_of_open-source_health_software#OpenAPS talk entry

MatthewVita, 2017-07-25:
I believe this issue can be closed because the remaining work will be done on the Wikipedia page.