openaps/openaps #125: Predict: negative IOB confuses predictions of BWP

kempera, 2017-05-28:
Great tool, but BWP comes at low BG-Level (e.g. 62) to wrong (& dangerous!!!) conclusion that negative IOB (e.g.-0.78U) effected by negative sensitivity (e.g. -65) will increase BG-level (expected effect: -0.78 x -65=--51 md/dl and expected outcome: 63--51=114 mg/dl). Theoretically correct, but creating wrong incentive to increase Basal rate to raise BG level... Am I missing something? Any suggestions to prevent a "downward Hypo spiral"?

kempera, 2017-05-28:
ps: sorry, expected outcome of example should be: 62--51=--113 mg/dl

scottleibrand, 2017-05-28:
Sorry, why do you say that is wrong? 

We don't use BWP (at all). With OpenAPS oref0, if you're 62 with -0.8U of IOB and a resulting eventualBG above target, it will most likely run a neutral basal as long as you're flat, and then start high-temping if you start rising faster than expected from the negative insulin (faster than BGI).