openaps/openaps #135: Are you still maintaining this code base?

nicored, 2018-02-09:
Hi,
I noticed that issues raised 2 years ago haven't been fixed yet. Are you still maintain this repo?

scottleibrand, 2018-02-09:
There is lots of activity still happening in oref0, but this repo, the openaps toolkit, is mostly frozen now. We’re actually working on moving from this code to @ecc1’s new Go based toolkit, which is much faster (and actively maintained). See the openaps/hardware-dev channel on Gitter and the go-explorer-hat branch of oref0.

pravarag, 2019-02-25:
Hi, are you planning to completely drop python-client for for OpenAPS ? I wanted to contribute to this project, so where shall I look into?

scottleibrand, 2019-03-10:
@pravarag yes, we plan to completely drop all of our unmaintained python dependencies eventually.  If you haven't already, I'd check out these PR and issues lists:
- https://github.com/openaps/oref0/pulls
- https://github.com/openaps/oref0/issues
and these gitter chat channels:
- https://gitter.im/nightscout/intend-to-bolus
- https://gitter.im/openaps/oref0
- https://gitter.im/openaps/hardware-dev
to get a pulse of what people are working on and what people need assistance with.

closed 2019-03-10 by scottleibrand