openaps/openaps #136: recurring 'Could not get subg_rfspy state or version' errors

rainbow-lazer, 2018-05-07:
I've been experiencing this on one of my boards over the last month or so running 0.6.1. Tried all the troubleshooting tips in the docs, as well as re-flashing a couple times. The fix lasts a few days at most, then back to a flaky radio  One thing I notice is that if the rig loses connection with the pump, it seems to throw it into the 'subg_rfspy state or version' error...

Per Facebook conversation with Katie DiSimone and Dana M. Lewis, I'm also attaching screenshots of oref0-runagain after installing 0.7.0-dev and answering the cron questions: [1](https://www.dropbox.com/s/u7zfgwf6uq28ljs/Screen%20Shot%202018-05-07%20at%202.50.03%20PM.png?dl=0) [2](https://www.dropbox.com/s/t53tqlqaucn5ndf/Screen%20Shot%202018-05-07%20at%202.50.26%20PM.png?dl=0)

Kdisimone, 2018-05-07:
@rainbow-lazer can you move this over to the GitHub.com/openaps/oref0 repo?  I think it will get more attention there.  Thanks

rainbow-lazer, 2018-05-07:
will do thanks!

closed 2018-05-07 by rainbow-lazer