openaps/openaps #142: Open APS Bolus wizard treatments don't appear in NS

old-square-eyes, 2019-06-20:
After some investigation, this appears to be an OpenAPS issue.

https://github.com/nightscout/cgm-remote-monitor/issues/4685

Seems Bolus Wizard and Easy bolus for 722 pumps lose their mmol/L decimal point.

old-square-eyes, 2019-06-20:
What's weird is the decimal point IS included down stream in NSClient.

scottleibrand, 2019-06-28:
I presume you're using oref0 master (0.6.x)?  I don't believe oref0 dev (0.7.0) is affected by this bug, as it uses new pump history processing libraries instead of `decocare`.

old-square-eyes, 2019-06-28:
Yeah using Master

scottleibrand, 2019-06-28:
Please try `dev` and let us know if that fixes it.

Dave-Meyers, 2019-09-15:
Do you know if this was resolved? I'm using Master 0.6.3 and seem to have this exact problem. Should I try dev?

old-square-eyes, 2019-09-15:
Doesn't seem to be happening to me any more on Dev. Though I'm on a 522 mmol/L instead of the 722 I was on. Not sure if that made a difference.

Dave-Meyers, 2019-09-15:
Thanks for that. I'll give Dev a try. BTW my pump is a 522.